SirLynix/obs-kinect #1: Clipping via world coordinates

ScottSWu, 2020-02-09:
Hi, thanks for writing this plugin!
Just wanted to drop an idea based on how I was using the kinect.

A neat feature would be to use the Kinect's accelerometer or do some floor plane detection to determine real world XYZ coordinates, then clip the image using those coordinates.

For example, you could remove the floor from the scene, or a wall.

SirLynix, 2020-02-09:
Hey, thank you for your suggestion!

I can see how to get the floor clip plane (using `IBodyFrame::get_FloorClipPlane`), which may require a calibration step (since the floor is only detected when the body feet are), and deriving the wall from there (just asking the user to set their distance), and sure, that could be useful. I'll keep it in mind. Thanks for the idea!

About the accelerometer, I just got no idea how to even access it from the Kinect SDK API :-/

SirLynix, 2020-12-05:
I started to use NuiSensorLib.lib (for KinectV2) which gives a lot more control and info about the sensor, including it's accelerometer.

Which means I now have access to the accelerometer value for every Kinect model!