SirLynix/obs-kinect #2: [v1] "Kinect Source" not available/present when adding source

Timmy-P, 2020-02-13:
Going to preface this with a "thank you" for making this plugin.  Very clever to be able to implement stuff like this into OBS to see all the creative ways it can be used.

That being said, after installing the plugin I'm unable to add the Kinect into my scene, as the source entry isn't showing up when I try to add a source.  Granted, I *am* using a Kinect v1 (in which I have to use versions 1.8 of the Drivers, SDK, and Toolkit), which might have something to do with it.  Will the source entry only show up when the program/plugin detects a Kinect is connected to the system, or if a certain condition is met?

SirLynix, 2020-02-14:
Hello, thanks for trying out this plugin.

If you can't even see the KinectSource available in OBS, it means the plugin failed to load (you should have something about it in OBS log file), which I think is caused by the driver version you're using. The plugin is linked against the Kinect SDK 2.0, which is probably not compatible with 1.8.

obs-kinect only does stuff with your Kinect when a KinectSource is created/visible (unless "Disable when invisible" checkbox is left unchecked).

A friend of mine tested it already with a Kinect v1 and it didn't work (even though the source was available), I suspect this is because for now I'm opening the device with Color/Depth/IR enabled buffers (afaik the Kinect v1 doesn't like outputting IR).

Are you familiar with OBS plugin compilation? If so, could you try disabling IR to see if it works for you? If not, could you install Kinect SDK 2.0 and test its examples? Do they all work?

Thank you.

Timmy-P, 2020-02-14:
Thank you for the response.

I didn't think to look at the logs, the plugin is indeed not loading at all:
```
14:49:25.962: LoadLibrary failed for '../../obs-plugins/64bit/obs-kinect.dll': The specified module could not be found.
14:49:25.962:  (126)
14:49:25.962: Module '../../obs-plugins/64bit/obs-kinect.dll' not loaded
```

This occurs even when I have tried installing the v2.0 variants of both the runtime and SDK, both on top of 1.8 and on its own (after uninstalling everything 1.8 related).  This might be something on my end, considering your friend was able to get the plugin to load and was able to add an empty Kinect Source.  Nonetheless, if the plugin is linked to only the SDK 2.0, then I'm going to have a bad time, haha.

The 2.0 variants of the drivers and software are exclusive to the v2 Kinects, both Xbone and Kinect for Windows v2.  Poking around the examples in the 2.0 SDK show that the software doesn't even see my v1 *at all,* which is unfortunate, to say the least.

I have very light experience with C++, but no experience with OBS plugins specifically (though I have quite a few tabs open that are hoping to change that.)  I'll see if I can cut out the IR, and/or redirect the plugin towards 1.8 and see what happens.

Thank you.

SirLynix, 2020-02-15:
Hmm, I think I'll need to make some dev to add Kinect v1 support, since it doesn't seem to work like v2 at all. Damn.

SirLynix, 2020-04-26:
I'm glad to announce Kinect v1 support is on its way! #6

SirLynix, 2020-05-01:
Some update about v1 (and obs-kinect 0.3) can be found in #7!

Timmy-P, 2020-05-01:
Nice, thank you so much for working on v1 support! I'll go ahead and close this and stay in #7 for now.

closed 2020-05-01 by Timmy-P