SirLynix/obs-kinect #3: Plugin not loaded

Rakohn, 2020-03-01:
Hi !

I'm trying to install your plugin but it is not loaded by OBS at startup.

Here's what I found in OBS logs : 
`17:57:37.963: LoadLibrary failed for '../../obs-plugins/64bit/obs-kinect.dll': The specified module could not be found.
17:57:37.963:  (126)
17:57:37.963: Module '../../obs-plugins/64bit/obs-kinect.dll' not loaded`

Plugin files are correctly copied, I'm using a kinect v2 with 2.2_1905 runtime library installed.

My goal is to use your plugin for mixed reaity with an oculus quest.

SirLynix, 2020-03-01:
Hello. This is weird, could you try installing the Kinect2 SDK? (from
https://www.microsoft.com/en-us/download/details.aspx?id=44561)

Maybe the plugin is linked to the SDK and should be linked to the runtime?

Rakohn, 2020-03-01:
I've installed the SDK, same issue :(

SirLynix, 2020-03-01:
I think I know what it is, could you please try to install visual studio
redistribuables?

32bits: https://aka.ms/vs/16/release/vc_redist.x86.exe
64bits: https://aka.ms/vs/16/release/vc_redist.x64.exe

If that still not works, please open the dll file using
http://www.dependencywalker.com to see what dependency you're missing?

Thanks

Rakohn, 2020-03-02:
Ok, it works now, that weird, I was pretty sure I was up to date with VC redist.

SirLynix, 2020-03-02:
Glad to hear that!

I hope this plugin will serve you well, feel free to create a new issue if
you experience any trouble or have an idea for a new feature.

closed 2020-03-02 by SirLynix