SirLynix/obs-kinect #6: v1 support?

metaColin, 2020-04-12:
Hey just curious what the state of v1 is? Also OS X?

I have some libfreenect experience. Lemme know if I can help.

JonathanFly, 2020-04-12:
I'm interested too. I have a bunch of Kinect 360s from an abandoned project. Would be fun for them to have a use. Anything I can test for you?

SirLynix, 2020-04-12:
Hi. Sure, the big problem I have with Kinect v1 support is that I ... lost it. I truly have no idea where I put my Kinect v1 and cannot test the support.

@metaColin: Well you can add support for Linux/macOS/KinectV1 by making your own KinectDevice class (it contains all code related to Kinect).

@JonathanFly: Sure, if I can't find my Kinect v1 it would be nice if you could test the plugin with it (I don't feel conformtable adding support without testing the device but I can do it).

Timmy-P, 2020-04-12:
@SirLynix 
I have a v1 and adapter, and would be more than happy to test out the plugin as well. Go ahead and do whatever is comfortable with you.

JonathanFly, 2020-04-12:
Kinect v1s are often <20 dollars on ebay. If it works well that would be pretty good value as an extra camera. Video quality is a lot lower than v1 but may still be useful.

metaColin, 2020-04-12:
I find v1s  at Goodwill for $10ish all the time (sometimes even with the usb adaptor).

The v1s are great for teaching because and they work with Processing and can be driven by a Raspberry Pi.

Newer depth cameras like the Azure Kinect Dk Developer Kit look sweet, but I can setup a teenager who is just learning about code with a Kinect v1 + a Raspberry Pi for under $100 dollars.

SirLynix, 2020-04-26:
I just found my old Kinect v1 by accident, and guess what? It turns out it's surprisingly working well! I'm glad to announce obs-kinect 0.3 will support Kinect v1 (and will be the biggest update this plugin ever got!).

metaColin, 2020-04-26:
Cool! I'll promise I'll put it through it's paces!

I have:
3 x Kinect v1
1 x Kinect v2

I wonder if my computer/obs can handle all 4 at once!

:-P

JonathanFly, 2020-04-26:
I actually have both versions as well.

Kinect 2 works well except for the lack of exposure control, it's always set to automatic. This means if you try to use a typical webcam setup where you light your face more than the background, it adjusts to see the background and blows out your face. 

Looks like it's possible to get at it: https://github.com/OpenKinect/libfreenect2/issues/1031

SirLynix, 2020-05-01:
Some update about v1 (and obs-kinect 0.3) can be found in #7!

SirLynix, 2020-05-13:
[obs-kinect 0.3](https://github.com/SirLynix/obs-kinect/releases) release candidate is now available, with Kinect v1 support. Please let me know if and how it works for you!

marv3lls, 2020-05-28:
@SirLynix it's working tremendously!

closed 2020-12-26 by SirLynix