SirLynix/obs-kinect #9: Freenect(2) update thread (Linux/macOS support)

SirLynix, 2020-05-03:
Hi,

With the upcoming 0.3 release, I splitted obs-kinect into multiple parts: common stuff like obs plugin registration, depth/infrared/faux green-screen processing, source and general device handling is part of the core but it doesn't know about Kinect devices (or even windows). Kinect v1/v2 handling is done in separate dll which are loaded by obs-kinect at startup.

This was made to add support for Kinect v1 without requiring the install of both v1 and v2 runtimes. And also to support Linux/macOS.

Obviously, Microsoft Kinect SDKs don't work on Linux/macOS, hopefully projects like [libfreenect](https://github.com/OpenKinect/libfreenect) and [libfreenect2](https://github.com/OpenKinect/libfreenect2) allow the use of Kinect on these systems.

Since obs-kinect device handling is done in separate libraries, it's possible to add libfreenect support kinda easily.

I tried both of these libraries on Windows (since they do have Windows support as well) for testing and it's working well!

**freenect (v1):**
[image]
[image]

**freenect2 (v2):**
[image]

However since libfreenect(2) needs raw USB commands, it requires to install a custom USB driver (like [UsbDk](https://github.com/daynix/UsbDk)). Hopefully this doesn't conflict with Kinect v2 (I can use libfreenect2 and Kinect SDK v2.0 without problem), but well, for Kinect v1 it's a bit more like hell.  You have to replace the Kinect USB driver (using Zadig) for libfreenect to work which prevents Kinect SDK v1.8 to work, and it took me one hour to reinstall the official Kinect drivers, so yeah, I don't think I will release an obs-kinect libfreenect backend on Windows because of that (since it doesn't seem to do things you cannot do with official Kinect v1.8 runtime, except maybe for led control?).

As for Kinect v2, libfreenect2 does add more control, especially with the RGB camera color settings (things like exposure mode, white balance, gain, and such), so I think I'll support it on Windows too (even though I hope to be able to do that as well with the official SDK backend, using [this](https://github.com/Microsoft/MixedRealityCompanionKit/tree/master/KinectIPD/NuiSensor)).

## Important note

Both libfreenect libs are really cool, but they are lacking some higher-level stuff the Kinect SDK does, like body/skeletal detection, or dedicated background removal (like Kinect SDK v1.8 does). It doesn't mean you won't be able to have background removal on Linux/macOS but it means you won't be able to use body informations for that and there's not much I can do about that. (except implementing body detecting but it requires some skill I don't have).

I will be able to compile the plugin on Linux, or even test it without problems, but I won't be able to do that with macOS. So if you have a mac and are willing to help, please let me know!

Also, as the 0.3 release is already a big one, Linux/macOS support won't be part of it (but I'll try to do it asap).

TL;DR: Linux/macOS support requires an open-source backend since official Microsoft SDKs are for Windows. There are backend for Kinect v1 and v2 but they don't do higher-level processing like body information. Things like body detection/greenscreen filtering won't work on Linux/macOS because of that.

skinlayers, 2020-05-06:
@SirLynix I can do Mac builds and testing for you. I have both a Kinect v2 and a v1.

SirLynix, 2020-05-06:
Thanks! I have little experience with macOS compilation, I'll probably need some help!

erdemthearch, 2020-05-07:
Hello :)
it seems that you only changed "include" and "src" files. 
So how can we update the software? Which files must be copied and pasted to where?
Thanks in advance.

SirLynix, 2020-05-07:
Hi! On which plateform are you?

Premake should handle all the config for tout.

erdemthearch, 2020-05-08:
W10 and Kinect v2

SirLynix, 2020-05-08:
If you're using visual studio community 2019 all you have to do is install
Kinect For Windows Runtime 2.0 and execute ’premake.exe 2019’, this will
generate a solution which you can build. See readme for more info

erdemthearch, 2020-05-08:
I am really sorry but i am not a coder. I only know node based coding. I am really far away from visual studio. Is there any quick istall file or something like that? Or do you consider to make one for end users like me in future? Thanks :)

SirLynix, 2020-05-08:
You can download the 0.2 release in the releases section of GitHub. The 0.3 binaries will be downloadable soon (I will make a release candidate for them).

As for the Kinect v2, there's not much difference between 0.2 and 0.3 for now, except for some performance improvement.

HolyStephano, 2020-12-14:
I've tested libfreenect on my current machine (archlinux install) and it works pretty well. Is there any reasonably straightforward way to build this with libfreenect as it currently stands? I'd be willing to go through the effort and test it on my system, this is something i've been wanting for a while now!

SirLynix, 2020-12-14:
Hi!

From what I saw libfreenect(2) should be kinda easy to use, and Linux support is my next big task. It's just that I have a lot of work to do for other projects (and I have a relatively costly hobby called eating 😄 ) and this is not at the top of my todo list for now.

However a lot of the work that went into 0.3 (that will be 1.0 when out) was to make obs-kinect core portable, and leave Windows/Kinect SDK specific stuff in some backend code. All that to say there's not much work remaining in order to have this plugin on Linux!

HolyStephano, 2020-12-14:
Fantastic! If you need someone to test, please feel free to find a way to contact me haha.

fotiDim, 2021-01-17:
I am trying to use Kinect 2 as a webcam on Mac (using Slack). Using Libfreenect2 I can get the RGB stream through their example app (Protonect) or code but there is no straightforward way to expose Kinect as a webcam to other software.

Since OBS can act a virtual webcam on Mac having this plugin working would be highly useful.

oixtron, 2021-01-27:
Hello!
I have to agree, I would love to see a working macOS Plugin. I've been looking for a solution to use my Kinect v2 via libfreenect2 as a webcam too for quite some time now, this is by far the most promising solution. Great work so far!

SirLynix, 2021-01-28:
Just a quick message to tell you guys I'm not forgetting about Linux and macOS, I'm just working on several things at once (a man has to eat). My next big task (and the only remaining missing feature for 1.0) is Linux and macOS support. 😄

SirLynix, 2021-01-31:
[image]

First KinectV2 color image captured using the WIP freenect2 background!
I'm testing it on Windows for now but this backend is made for Linux and macOS users 😄 

I'm also switching from premake to xmake as a build tool for its handling of dependencies, and planning to setup a CI (with nightly builds).

SirLynix, 2021-02-02:
[image]
I managed to make the fake greenscreen work with libfreenect2. The results seems worse than with the official Windows SDK (it may be because of some options I still have to tweak).

fotiDim, 2021-02-02:
That is cool. Personally I am happy with having greenscreen only on Windows. I only need it as a webcam on mac.

MarcoWilli, 2021-02-08:
Do you know if the Linux version will have the same impact on system performance than on Windows? I dropped the Kinect atm on Windows because of that. 

I hope it will be less, as at least the Kinect windows service will not be there?! 

Otherwise I will sell my Kinect due to lack of availability of graphics cards at normal prices atm

SirLynix, 2021-02-08:
It depends on the Kinect model.

One of the benefit of Freenect2 (kinect for xbox one) is that it is able to exploit either the CPU or GPU to perform depth processing. I don't know if it will be faster than what the official SDK does but at least there will be more control.

I don't really know about Freenect (kinect for 360), as it's a completely different API, but I think it will be handled in a smarter way (the official Windows SDK tries to use **all** CPU threads at once).

As for Azure Kinect, I don't think there will be a significant difference (as it's the same SDK). Even though there maybe a difference between Windows and Linux/macOS on low-level handling.

So the real answer here is: I don't know yet, it requires some testing.

SirLynix, 2021-02-11:
I managed to setup a CI which will build Linux plugins files (and eventually Windows and macOS) with a preliminary support for Kinect v2 (xbox one) through freenect2, and Azure Kinect support through Kinect for Azure SDK.

If you're feeling adventurous, you can download the Linux plugins files [here](https://github.com/SirLynix/obs-kinect/actions?query=workflow%3ALinux-Build) (click on the last workflow that ran on master and download the ubuntu-latest-x64-releasedbg artifacts).

You'll still need a few other files (currently missing from this package), so you'll need to download the [latest Windows nightlies](https://github.com/SirLynix/obs-kinect/files/5743062/obs-kinect-locales.zip) to get theses files (please do it only if you know what you're doing).

**This is not tested.**

If you went into the troubles of trying this, please let me know if and how it worked for you (even if it didn't)!

Kinect v1 (xbox 360) and macOS users, don't worry. You'll be able to try it out soon!

SirLynix, 2021-02-13:
I improved packaging support for Windows and Linux, which means the Linux package now has obs-kinect data (shaders and locales).

I'm wondering, what would be the best way to distribute obs-kinect on Linux? I've seen that some plugins (such as [obs-websocket](https://github.com/Palakis/obs-websocket/releases/tag/4.9.0)) build a .deb package which automatically installs the plugin files at the right place, which is something I can do, but doesn't that excludes non-debian based Linux distributions? What about them? Is obs studio even available/used on such distributions?

About mac builds, things seems easier on a technical level but would require me to pay for an Apple Developer Certificate annually. I need to think about this, and if there's another way.

fotiDim, 2021-02-13:
I believe I can help with Apple’s developer account. I think I can add you
as a member under my account.

MarcoWilli, 2021-02-13:
> I'm wondering, what would be the best way to distribute obs-kinect on Linux? I've seen that some plugins (such as [obs-websocket](https://github.com/Palakis/obs-websocket/releases/tag/4.9.0)) build [...]
 
Yes, if it´s a .deb package it´s only for debian based distros. AFAIK advanced users could extract that package and copy to the right places on their distro, too. IIRC I´ve done that once but not sure. I tried Solus, elementaryOS and am now with Manjaro (arch based). For all theses distros there is OBS "natively" available through their package managers.

Compiling it for one/serveral formats would limit it to these distros. Maybe at least give a manual how to install from scratch into most common distros/bases (don´t know the right term) (deb, arch). There are also distro-wide packages like flat, snap and some third I can´t remember. Though I´m not sure if this will work only together with OBS installed using this packages, too.

SirLynix, 2021-02-17:
First image of obs-kinect on Linux!
[image]

I installed Ubuntu 20.04 and went into some troubles but finally found a way to make it work.

From what I observed, freenect2 works well, and is able to use my KinectV2 without any issue, except for the root permissions (which seems to be easy to fix, see https://github.com/OpenKinect/libfreenect2/issues/870 ).

The Azure backend works too, but for some weird reason I couldn't retrieve the depth or infrared streams, probably something on my side.

The main remaining issue which remains is now a rpath one, I'll explain:
OBS-Studio loads obs-kinect which then loads its backends (obs-kinect-freenect2 for example), but this one is linked against obs-kinect and it seems Linux wants to open libobs-kinect.so back from there, except it just can't be found since libobs-kinect belongs to the plugin folder.

Setting the rpath to $origin doesn't help, as the origin here is OBS, and I can't set it to something like `$origin/../../obs-plugins/64bit` (which I did to test locally) because this path isn't guaranteed.

I see some solutions to fix this:
1) Find some rpath trick to retrieve libobs-kinect.so.
2) Split libobs-kinect.so in two files, set one next to obs executable and the other in obs-plugins folder (and make the backend linked to the first one, so no direct link between the plugin and its backend).
3) Use static linking.
4) Merge obs-kinect backend code into the main plugin and dynamically load backend dependencies.

I'm not sure if 1) is even possible, 3) is ugly and 4) is difficult to implement (as the libraries used are not really meant for that), so I think I'll go with 2).

Oh and I had to fix a shader on Linux, which now doesn't compile on Windows, yay.

SirLynix, 2021-02-18:
I managed to work around theses problems and even make my Azure Kinect work on Linux (with depth-based blur!), I was just lacking Microsoft proprietary depth engine.

If the last commit compiles on the CI, it should be working for Linux users.

[image]

HolyStephano, 2021-02-21:
> I'm wondering, what would be the best way to distribute obs-kinect on Linux? I've seen that some plugins (such as [obs-websocket](https://github.com/Palakis/obs-websocket/releases/tag/4.9.0)) build [...]

obs studio is available on most mainstream distros (debian, arch, gentoo probably even) and your "duty" as the developer in the typical linux packaging world is just really just tagging releases in github. Then those builds are reproducible for package maintainers in theory.

SirLynix, 2021-02-21:
Do you know a OBS Studio plugins for Linux which does everything that should be done for this platform? The only example I've seen is obs-websocket which only provides a .deb package for Debian-based distributions.

MarcoWilli, 2021-02-21:
On my previous Manjaro installation I had the "NDI" plugin installed. Oh and on my laptop with Solus OS, too.
Don't know if this helps you.

SirLynix, 2021-02-21:
How did you install them? I can only find debian packages on their [Github page](https://github.com/Palakis/obs-ndi/releases)

MarcoWilli, 2021-02-21:
Honestly I can't remember. I just did a lookup and at least on Manjaro there are packages available in the AUR from where it can be build on demand. For Solus it COULD be similar.

But iirc I had to adapt some files/paths at least on one of the two OS. Though I'm really really not sure about. I do not use OBS on Linux atm so I'm sorry I can not give you more information about it.

[image]

MarcoWilli, 2021-02-21:
I'm pretty sure though, that it depends on how one installs OBS.
Theses app/snap/flat packages are (all?) sandboxed and therefore plugins may be not compatible/different to install.

SirLynix, 2021-02-21:
Thanks for the info. I think I will just bother distributing a .deb package for now (once Linux support is complete) and we'll see for other distributions after that.

oixtron, 2021-07-17:
Any updates for the macOS users?

SirLynix, 2021-07-17:
Hi.

In theory obs-kinect master should already support macOS, but I think I need an Apple computer to build it myself, so I should try to setup a CI action for that.

Sorry for the wait, it's because of my lack of experience with mac platform.

SirLynix, 2021-07-20:
[image]
First image of the Freenect backend (Kinect v1 for Linux/macOS), it's happening!

xurei, 2021-09-05:
Is there a branch that we can test already ?

SirLynix, 2021-09-05:
Hi!
You can either build the plugin yourself or download the [latest Linux nightlies](https://github.com/SirLynix/obs-kinect/actions/workflows/linux-build.yml).

Freenect2 (Kinectv2) and Azure (Kinectv3) support should be fine, Freenect (Kinectv1) should work for color. It's a bit difficult for me to work on it at the moment but I'd love to have feedbacks.

xurei, 2021-09-05:
I just tried the plugin.
It does not seem to work, but I'm not sure it's the plugin fault.
For context : I have a Kinect v1 (model 1414). It is not detected by the plugin, or anywhere else except freenect-glview actually...

Here is what lsusb gives me :
```
Bus 001 Device 107: ID 045e:02ae Microsoft Corp. Xbox NUI Camera
Bus 001 Device 105: ID 045e:02b0 Microsoft Corp. Xbox NUI Motor
Bus 001 Device 106: ID 045e:02ad Microsoft Corp. Xbox NUI Audio
```

If you need help to test or debug anything, I'm here to help :-) I never developed a plugin for OBS, though.

SirLynix, 2021-09-05:
I think I might know what causes this, I'll try to commit a fix soon!

jekkos, 2021-12-03:
Same issue here, I installed the .so files in `/usr/lib/obs-plugins` and see that they are being loaded by the obs-studio. I got some errors on missing libraries which I added all to the folder. But no option to select the kinect pops up. I suppose this might be due to udev rules that I now just added.
Does anyone know how I can check if the plugin is installed fine? Also the build artifacts were removed by now.

SirLynix, 2021-12-03:
Check your OBS logs to see if the plugin is correctly loaded.

I just triggered a new build, it should be good in about 10min

jekkos, 2021-12-03:
I see no mention of the plugin files in the logs, however I do see some attempt in strace to load the .so files. 

Can they all go in the same dir or do I need to keep the subfolder structure?

jekkos, 2021-12-03:
currently I have obs-kinect-freenect.so, libobs-kinectcore.so and obs-kinect-freenect.so in `/usr/lib/obs-plugins`.

SirLynix, 2021-12-03:
Honestly I'm not sure at all of where obs will fetch plugins file, I did make it work one time on Ubuntu but I'm not using Linux much, especially not graphical Linux.

Build failed because of some link order issue, which is weird, I reported it to xmake author.

jekkos, 2021-12-03:
Ah ok, just wanted to know whether the .so files rely on the directory structure that comes with the zip file. But basically if the plugin is loaded I should be able to add it as a source even if there is no kinect connected?

SirLynix, 2021-12-03:
Yes, if the plugins loads you should be able to create a Kinect source, even if it failed to load any backend.

If you don't see a Kinect source in the list, then the plugin itself failed to load.

SirLynix, 2021-12-04:
Thanks to xmake author, the bug was quickly fixed and artifacts are now available for Linux.

xurei, 2021-12-06:
I tried the latest build and I still don't see the Kinect v1 in the devices list.
However, I can use the built-in video capture of OBS to use the Kinect as a camera - without much configuration options.

Hence, maybe this is why the plugin does not detect the Kinect ? (I actually doubt it, but if it can help you find the problem...)

SirLynix, 2021-12-06:
This seems pretty normal and isn't an issue, do you have any mention to obs-kinect in OBS logs?

b1tfl0w, 2021-12-24:
hello, i'm trying to run this plugin on Ubuntu 20.04 focal, with libfreenect installed from repositories and obs from ppa
```
freenect/focal,now 1:0.5.3-2 amd64 [installato]
  libreria per accedere ai dispositivi Kinect - metapacchetto

libfreenect-bin/focal,now 1:0.5.3-2 amd64 [installato, automatico]
  library for accessing Kinect device -- utilities and samples

libfreenect-demos/focal 1:0.5.3-2 amd64
  library for accessing Kinect device -- dummy package

libfreenect-dev/focal,now 1:0.5.3-2 amd64 [installato, automatico]
  library for accessing Kinect device -- development files

libfreenect-doc/focal,focal,now 1:0.5.3-2 all [installato, automatico]
  library for accessing Kinect device -- documentation

libfreenect0.5/focal,now 1:0.5.3-2 amd64 [installato, automatico]
  library for accessing Kinect device

obs-studio/focal,now 27.1.3-0obsproject1~focal amd64 [installato]
  OBS Studio for Ubuntu

```
then downloaded the releasedbg from here `https://github.com/SirLynix/obs-kinect/actions/runs/1539175629`

and put file using thi folder structure:

```
tree -L 3 .config/obs-studio/                                                                                                                                                 
.config/obs-studio/                                                                                                                                                                            
├── basic                                                                                                                                                                                      
│   ├── profiles
│   └── scenes
├── global.ini
├── logs
├── plugin_config
│   ├── rtmp-services
│   │   ├── meta.json
│   │   ├── package.json
│   │   ├── services.json
│   │   └── twitch_ingests.json
│   └── text-freetype2
├── plugins
│   └── obs-kinect
│       ├── alpha_mask.effect
│       ├── color_multiplier.effect
│       ├── gaussian_blur.effect
│       ├── greenscreen_filter.effect
│       ├── locale
│       ├── texture_lerp.effect
│       └── visibility_mask.effect
└── profiler_data

```
and:

```
tree -L 3 /usr/lib/obs-plugins

/usr/lib/obs-plugins/
├── cef_100_percent.pak
├── cef_200_percent.pak
├── cef_extensions.pak
├── cef.pak
├── chrome-sandbox
├── decklink-captions.so
├── decklink-ouput-ui.so
├── devtools_resources.pak
├── frontend-tools.so
[... 187 lines cut ...]
Attempted path: share/obs/obs-studio/images/overflow.png
Attempted path: /usr/share/obs/obs-studio/images/overflow.png
ALSA lib confmisc.c:767:(parse_card) cannot find card 'Camera'
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_card_driver returned error: No such device
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_concat returned error: No such device
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_refer returned error: No such device
ALSA lib conf.c:5220:(snd_config_expand) Evaluate error: No such device
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM front:CARD=Camera,DEV=0
error: alsa-input: Failed to open 'front:CARD=Camera,DEV=0': No such device
ALSA lib confmisc.c:767:(parse_card) cannot find card 'Camera'
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_card_driver returned error: No such device
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_concat returned error: No such device
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_refer returned error: No such device
ALSA lib conf.c:5220:(snd_config_expand) Evaluate error: No such device
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM front:CARD=Camera,DEV=0
error: alsa-input: Failed to open 'front:CARD=Camera,DEV=0': No such device
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to create effect: shader error
Aborted (core dumped)

```

SirLynix, 2021-12-24:
It looks like obs-kinect-freenect.so, and locale/shader files are not where they should be. If I recall correctly obs-kinect-freenect.so should be next to obs executable, dunno for locale/shaders files.

b1tfl0w, 2021-12-25:
osb executable is in `/usr/bin` but dyinamically linked libraries `.so` usually go in `/usr/lib`

b1tfl0w, 2021-12-25:
Ok now the kinect window opens atlist, but kinect is not detected...this is what I have:

```
tree -L 3 /usr/share/obs/obs-plugins/obs-kinect/

/usr/share/obs/obs-plugins/obs-kinect/
├── alpha_mask.effect
├── color_multiplier.effect
├── gaussian_blur.effect
├── greenscreen_filter.effect
├── locale
│   ├── cs-CZ.ini
│   ├── de-DE.ini
│   ├── en-US.ini
│   ├── fr-FR.ini
│   ├── ko-KR.ini
│   └── pl-PL.ini
├── texture_lerp.effect
└── visibility_mask.effect
```

```
/usr/lib/obs-plugins/
├── libobs-kinectcore.so
├── obs-kinect-azuresdk.so
├── obs-kinect-freenect2.so
├── obs-kinect-freenect.so
├── obs-kinect.so

```
[image]

```
cat obstrace.txt | grep kinect
stat("/usr//lib/obs-plugins/libobs-kinectcore.so", {st_mode=S_IFREG|0664, st_size=51960, ...}) = 0
stat("/usr//lib/obs-plugins/obs-kinect-azuresdk.so", {st_mode=S_IFREG|0664, st_size=48096, ...}) = 0
stat("/usr//lib/obs-plugins/obs-kinect-freenect2.so", {st_mode=S_IFREG|0664, st_size=1347992, ...}) = 0
stat("/usr//lib/obs-plugins/obs-kinect-freenect.so", {st_mode=S_IFREG|0664, st_size=292136, ...}) = 0
stat("/usr//lib/obs-plugins/obs-kinect.so", {st_mode=S_IFREG|0664, st_size=97720, ...}) = 0
access("/usr/share/obs/obs-plugins/libobs-kinectcore", F_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr//lib/obs-plugins/libobs-kinectcore.so", O_RDONLY|O_CLOEXEC) = 24
access("/usr/share/obs/obs-plugins/obs-kinect-azuresdk", F_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr//lib/obs-plugins/obs-kinect-azuresdk.so", O_RDONLY|O_CLOEXEC) = 25
access("/usr/share/obs/obs-plugins/obs-kinect-freenect2", F_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr//lib/obs-plugins/obs-kinect-freenect2.so", O_RDONLY|O_CLOEXEC) = 25
openat(AT_FDCWD, "/usr//lib/obs-plugins/libobs-kinectcore.so", O_RDONLY|O_CLOEXEC) = 25
access("/usr/share/obs/obs-plugins/obs-kinect-freenect", F_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr//lib/obs-plugins/obs-kinect-freenect.so", O_RDONLY|O_CLOEXEC) = 25
access("/usr/share/obs/obs-plugins/obs-kinect", F_OK) = 0
openat(AT_FDCWD, "/usr//lib/obs-plugins/obs-kinect.so", O_RDONLY|O_CLOEXEC) = 25
access("/usr/share/obs/obs-plugins/obs-kinect/locale/en-US.ini", F_OK) = -1 EACCES (Permission denied)
openat(AT_FDCWD, "./obs-kinect-azuresdk.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
write(2, "error: os_dlopen(./obs-kinect-az"..., 152) = 152
openat(AT_FDCWD, "./obs-kinect-freenect.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
write(2, "error: os_dlopen(./obs-kinect-fr"..., 152) = 152
openat(AT_FDCWD, "./obs-kinect-freenect2.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
write(2, "error: os_dlopen(./obs-kinect-fr"..., 155) = 155
openat(AT_FDCWD, "./obs-kinect-sdk10.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
write(2, "error: os_dlopen(./obs-kinect-sd"..., 143) = 143
openat(AT_FDCWD, "./obs-kinect-sdk20.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
write(2, "error: os_dlopen(./obs-kinect-sd"..., 143) = 143
write(1, "info:     obs-kinect.so\n", 24) = 24
write(15, "02:44:02.168:     obs-kinect.so\n", 32) = 32
access("/usr/share/obs/obs-plugins/obs-kinect/color_multiplier.effect", F_OK) = 0
openat(AT_FDCWD, "/usr/share/obs/obs-plugins/obs-kinect/color_multiplier.effect", O_RDONLY) = 34
access("/usr/share/obs/obs-plugins/obs-kinect/gaussian_blur.effect", F_OK) = 0
openat(AT_FDCWD, "/usr/share/obs/obs-plugins/obs-kinect/gaussian_blur.effect", O_RDONLY) = 34
access("/usr/share/obs/obs-plugins/obs-kinect/greenscreen_filter.effect", F_OK) = 0
openat(AT_FDCWD, "/usr/share/obs/obs-plugins/obs-kinect/greenscreen_filter.effect", O_RDONLY) = 34
access("/usr/share/obs/obs-plugins/obs-kinect/gaussian_blur.effect", F_OK) = 0
access("/usr/share/obs/obs-plugins/obs-kinect/texture_lerp.effect", F_OK) = 0
openat(AT_FDCWD, "/usr/share/obs/obs-plugins/obs-kinect/texture_lerp.effect", O_RDONLY) = 34
access("/usr/share/obs/obs-plugins/obs-kinect/texture_lerp.effect", F_OK) = 0
access("/usr/share/obs/obs-plugins/obs-kinect/visibility_mask.effect", F_OK) = 0
openat(AT_FDCWD, "/usr/share/obs/obs-plugins/obs-kinect/visibility_mask.effect", O_RDONLY) = 34
access("/usr/share/obs/obs-plugins/obs-kinect/alpha_mask.effect", F_OK) = 0
openat(AT_FDCWD, "/usr/share/obs/obs-plugins/obs-kinect/alpha_mask.effect", O_RDONLY) = 34

```

SirLynix, 2021-12-25:
It looks like your locales are still not at the right place, same goes for libobs-kinectcore.so and obs-kinect-freenect(2).so, which if I recall correctly should be next to obs executable (I'm sorry it's been a while since I tested this on Linux).

jekkos, 2021-12-27:
@b1tfl0w I have gotten the libfreenect to load by starting obs from the `/usr/lib/obs-plugins` working dir. It was otherwise complaining with following message

```
14680 write(2, "error: os_dlopen(./obs-kinect-fr"..., 152error: os_dlopen(./obs-kinect-freenect->./obs-kinect-freenect.so): ./obs-kinect-freenect.so: cannot open shared object file: No such file or directory
```

while in the logging it did seem to pick it up fine before.

SirLynix, 2021-12-27:
Sorry about this guys, I honestly don't know how to handle this loading properly on OBS. As far as I recall obs-kinect.so (plugin file) should be in the obs-plugins folder and other .so should be next to obs executable (because they're loaded with a relative path).

Maybe loading them with a normal path and putting them in /usr/local/lib would be better.

jekkos, 2021-12-27:
indeed so you can also start obs from the .so folder as it will then pick up the files with the relative path from there. I am able to see color / depth views now using libfreenect, the background blur is also working when I use depth.

But when I switch to color the greenscreen stops working. This seems to be a libfreenect issue as I see quite some messages like the following 

```
info: [obs-kinect] [freenect] freenect notice: [Stream 70] Lost too many packets, resyncing...                                                                                                                                                                                                                                
                                                                                                                                                                                                                                                                                                                              
warning: [obs-kinect] [freenect] freenect warning: [Stream 70] Expected max 1748 data bytes, but got 1908. Dropping...                                                                                                                                                                                                        
                                                                                                                                                                                                                                                                                                                              
info: [obs-kinect] [freenect] freenect notice: [Stream 70] Invalid magic ffff                                                                                                                                                                                                                                                 
                                                                                                                                                                                                                                                                                                                              
info: [obs-kinect] [freenect] freenect notice: [Stream 70] Invalid magic ffff                                                                                                                                                                                                                                                 
                                                                                                                                                                                                                                                                                                                              
info: [obs-kinect] [freenect] freenect notice: [Stream 70] Invalid magic ffff    
```

This could be a local USB bandwith issue, I'm looking into this and trying to recompile libfreenect with some optimizations now

jekkos, 2021-12-27:
Ok I'm not sure if this is the issue for the greenscreen not showing in color mode. The messages also appear if I open the depth camera. I can also open both color/depth side by side, but alas no greenscreen on the color version.

SirLynix, 2021-12-27:
This is on me, the obs-kinect-freenect backend doesn't output color-to-depth info yet, mostly because it's a bit difficult to do with libfreenect (which seems to only provide the opposite: depth-to-color). I'll try to resume working on this backend for you 🙂

jekkos, 2021-12-27:
I didn't really know whether this was a bug on my side, but if it's a feature request, I'd be willing to add sth to your sponsor dashboard if you get it working ;)

[82 more replies]