SirLynix/obs-kinect #10: LoadLibrary failed with Kinect v1

saphir1997, 2020-05-12:
I have just added the new plugin version to OBS, but OBS will not start with the following messages:
`18:04:50.980: LoadLibrary failed for '../../obs-plugins/64bit/obs-kinect-sdk20.dll': (null) (126)`
`18:04:50.980: Module '../../obs-plugins/64bit/obs-kinect-sdk20.dll' not loaded`
`18:04:50.981: LoadLibrary failed for 'obs-kinect-sdk20': (null) (126)`
`18:04:51.198: Required value 'get_name' for '(null)' not found.  obs_register_source failed.`

At first glance it seems that it tries to load Kinect v2, even though I have just a Kinect v1.

I suspect this could be either that I do not have a Kinect v2 hooked up or that I have not installed the v2 SDK.

Did you ever try to start OBS with just a Kinect v1 attached?

By the way: Great work. Reviving this old hardware in such an amazing fashion.

SirLynix, 2020-05-12:
Hello,

It's a normal thing that obs-kinect tries to load both backends, as I cannot detect if any Kinect is attached without them, but it's not an error if it cannot load one (because of a missing dependency).

It seems what prevents obs-kinect to load is this line:
```
18:04:51.198: Required value 'get_name' for '(null)' not found. obs_register_source failed.
```

I suspect you're using OBS Studio 24.0, I compiled the plugin against OBS Studio 25.0, can you confirm that? Is upgrading an option for you?

saphir1997, 2020-05-12:
Yes, updating OBS from 24.0 to 25.0 indeed fixed the issue.
It is now working as expected. I would not have guessed, that this was the issue with this error log.
Thank you for your quick response.

closed 2020-05-12 by saphir1997