SirLynix/obs-kinect #13: [0.3rc1] Kinect v1 No image, "failed to open color stream: Unknown error" when enabling faux greenscreen

Timmy-P, 2020-05-13:
Going to link the logs upfront: https://obsproject.com/logs/93eDn7cI3SDKJ-gB
https://obsproject.com/logs/iwPUGmC7PA67g_vw

0.3 has been an absolute godsend to get the 360 Kinects up and running. It appears checking "Enable faux greenscreen effect" will result in no image showing, regardless of what filter type is used (I'm taking .greenscreentype_dedicated is to be used for V1).  Looking into the log gives the error "failed to open color stream: Unknown error" for every instance it is ticked; the color stream itself works, however, as shown below.
https://imgur.com/tXp4DUS

I double-checked everything needed in terms of runtimes for both Visual 2019 and Kinect, is there something else that's needed? I have the runtimes and SDKs for both 1.8 and 2.0, would it be possible those are conflicting?

roblevesque, 2020-05-13:
I also get this behavior. Good color stream with no green screen applied however same error as above  after choosing a filter method. 

I can also confirm that having only one SDK installed doesn't matter as I felt no need to install the SDK for v2 as I do not own one. I did however install it when the filter failed to function. 

Thanks!

SirLynix, 2020-05-13:
Hello. Sorry to hear that!

Could you guys try to put two Kinect sources in your scene (on the same device), one with color and one with depth stream, without enabling greenscreen.

This is to ensure it can fetch both stream at the same time, if that works then green screen with depth type should too. (if not, that's really weird).

If that doesn't work, could you install [Kinect for Windows Developer Toolkit v1.8](https://www.microsoft.com/en-us/download/details.aspx?id=40276), go to "Samples: C++" and try to run some of the examples (like Depth Basics, Depth with Color and such).

The biggest issue I have here is that I have no clue what the problem is, I'll work on a way to print the string representation of that error (that part seems broken) and release 0.3-rc2 tomorrow.

delet-1, 2020-05-13:
I have the same issue. Any option related to the depth stream does not work. In the developer toolkit demos I have no problem.

Maybe it has something to do with the differences between the Xbox 360 and Windows versions of the v1 Kinect. I can't test it because I only own the 360 version so it's just a guess.

Timmy-P, 2020-05-14:
Running depth gives the exact same issue, exactly: no image, "failed to open color stream: Unknown error." Here's the log:
https://obsproject.com/logs/wUICbRqBVv6rpc4_

Now, I do indeed have the toolkit installed and can run depth fine, but not their own background tool. I'll go ahead and do some fresh reinstallations of some of this stuff tonight and see if that helps.

delet-1, 2020-05-14:
@Timmy-P Which Kinect v1 model do you have?

Timmy-P, 2020-05-14:
> 
> 
> @Timmy-P Which Kinect v1 model do you have?

Don't know what you mean, it's the 360 version. Unless you mean if it's console vs. Kinect for Windows, in which it's console.

SirLynix, 2020-05-14:
Kinect v1 and v2 are very different and may have their own bugs (and to be honest, Kinect v1 is such a pain in the ass to use compared to v2).

Okay so if Microsoft examples are running it means the problem is on the plugin side, which is good. Also the fact that all of you have the issue is good, better than a random bug happening here and there without reason.

Here's RC2(-ish):
[obs-kinect-0.3-rc2.zip](https://github.com/SirLynix/obs-kinect/files/4627228/obs-kinect-0.3-rc2.zip)

It fixes translation of sources/greenscreen types, KinectSdk10 error messages and german translations.
It still won't work but at least we should get why in OBS logs.

delet-1, 2020-05-14:
This is the error I get when trying to access depth features with 0,3-rc2

[obs-kinect] failed to open color stream: The requested feateure is not available on this version of the hardware

SirLynix, 2020-05-14:
That spelling of "feature" tho.

Alright, so it seems my Kinect v1 is more recent and supports .. something which isn't support by yours.

I just noticed the error "failed to open color stream" is what is logged when you try to open the depth stream. So I'll assume it's the depth stream.

There are two things which may cause this I think: either the 640x480 depth resolution I request (instead of 320x240), either the near mode isn't supported.

Here are two builds, one with 320x240 res + near mode, one with 640x480 without near mode.
[obs-kinect-0.3-rc2-320x240.zip](https://github.com/SirLynix/obs-kinect/files/4627349/obs-kinect-0.3-rc2-320x240.zip)
[obs-kinect-0.3-rc2-without-near-mode.zip](https://github.com/SirLynix/obs-kinect/files/4627350/obs-kinect-0.3-rc2-without-near-mode.zip)

Could you try both of them to see if one is working?

Also could you try to run "Coordinate Mapping Basics-D2D" Kinect toolkit demo and enable near mode? Does it make a difference? (If you don't see yourself, try standing a bit further from the Kinect).

I hope the problem comes from the high-res depth map I'm asking from the Kinect, as it won't make much differences to disable it.

delet-1, 2020-05-14:
With 'obs-kinect-0.3-rc2-without-near-mode' it works correctly. 

The v1 has two models, the Xbox 360 version and the Windows version. The 360 version does not support Near Mode so I guess that was the problem. Thanks.

SirLynix, 2020-05-14:
Does everything work correctly? Even dedicated faux green-screen?
Have you tried Microsoft example? Does it have a near mode checkbox? Does checking it changes anything?

Sorry to harass you with questions but that will help me fix this for good.

delet-1, 2020-05-14:
Yes, every type of faux greenscreen work including KinectSDK background removal option.

In the Microsoft examples I can check the Near Mode option but it has no effect.

SirLynix, 2020-05-14:
https://docs.microsoft.com/en-us/archive/blogs/kinectforwindows/kinect-for-windows-building-the-future
http://thingx.cloud/wp-content/uploads/2012/08/near-mode-what-it-is-and-isn-t.aspx

Well, it seems indeed that near mode was introduced with Kinect for Windows. I'm not sure whether it's a hardware change or a firmware update.

I will introduce the near-mode checkbox in the real rc2, that should make it compatible with Kinect for 360. Sorry you guys cannot use near mode.

delet-1, 2020-05-14:
Glad to be able to help!

SirLynix, 2020-05-14:
Fixed by [v0.3-rc2](https://github.com/SirLynix/obs-kinect/releases).

closed 2020-05-14 by SirLynix