SirLynix/obs-kinect #16: Standalone Driver

sew3521, 2020-05-18:
Would you be able to make a standalone driver to allow for the Kinect to be used as a webcam? Currently this is able to happen with OBS-Virtual Cam.
With the shortage of webcams right now I bet you would be able to sell something like that for a few bucks. I know I would be willing to spend the money so I didn't have to run OBS while on a Skype call.

SirLynix, 2020-05-18:
Hi! Thanks for the idea.

The problem here is I have zero experience with webcam drivers, so I'd have to spend a lot of time learning how to do this to have a basic support for Kinect v1/v2 color streams (as the greenscreen stuff relies a lot on the GPU, this complicates a lot of stuff).

Also, Kinectv2 is already recognized as a webcam by Windows, once you've got the runtime installed. So this will only be a thing for Kinect for 360/Windows (v1), I wonder if there are really a lot of people here with this need.

But hey, this is a great idea, I can't say I won't look at that in the future, maybe. :)

sew3521, 2020-05-18:
That is odd. I cannot see my Kinect2 as a webcam inside of Teams or Skype. I have been using your awesome plugin plus OBS. Thanks for the awesome plugin

SirLynix, 2020-05-18:
Oh really? Maybe that requires the Kinect 2.2 SDK. I don't have Teams nor Skype but OBS / Discord / Facebook are all able to use my Kinect v2 as a webcam so I think it's supported system-wide on my computer.

Maybe that webcam driver idea may help more people than I though.

SirLynix, 2020-05-18:
I'm curious btw, are you only using the color stream of the Kinect or are you using the greenscreen effect as well?

sew3521, 2020-05-18:
I know OBS does not see it as a webcam. I did not try Discord or Facebook but can try them later tonight. When I was looking around I saw someone say that 64 bit versions of programs cannot see it as a webcam but 32 bit versions can. I have not tested that but I will look into that as well.
I am only using the color stream. The green screen idea is a brilliant idea however I do not have a need for it.

closed 2020-12-26 by SirLynix