SirLynix/obs-kinect #19: Lagging/stuttering

genregasm, 2020-05-26:
Machine details:
Ryzen 9 3900x
GeForce 2080
32gb RAM
Everything on SSDs
Asrock x570 Steel Legend
Windows 10 with all latest updates, recommended visual studio installs, kinect runtime
Kinect Configuration Verifier shows all green checks except USB, shows yellow (!) due to non-Intel or Realense USB 3.0 ports, but works fine

When I use Kinect with depth sensor in OBS, it stutters and lags. This doesn't happen when I use the Kinect in TouchDesigner (or any other program), in fact, it doesn't happen when I use TouchDesigner to Spout to OBS, so I have narrowed down that this is an issue with the plugin for OBS itself. Regular camera via the plugin works fine, it's just anything using the depth sensor lags.

Timmy-P, 2020-05-26:
Is OBS up to date? Are you running OBS as administrator? What's your system usage when using depth?

SirLynix, 2020-05-26:
The last person who had lag issues had USB 3 problems.

Are you using a Kinect 2? Is it connected to an USB 3 port? 

I don't know what TouchDesigner does, does it uses the depth camera?

How are you using the depth camera? Greenscreen or depth stream?

Are you using the 0.2 or 0.3 RC?

Also, could you install Kinect for Windows SDK 2.0 (or 1.8 if you're using a Kinect v1) and try to run some examples? Are there working well?

genregasm, 2020-05-30:
I am that last person, and have determined it is not a USB 3 problem since
it works fine on everything else.
Kinect v2, USB 3.1 port, all drivers up to date. TouchDesigner uses depth
camera. OBS plugin works fine using normal Kinect cam, but only stutters
using greenscreen mode. I have Kinect SDK 2.2 I believe (latest version).
What is RC?

SirLynix, 2020-05-30:
RC is Release Candidate, 0.3 is not yet out but is available for testing (in the [releases](https://github.com/SirLynix/obs-kinect/releases) section here). Are you using that version?
Do Microsoft SDK examples work fine?

genregasm, 2020-06-01:
Yes, I updated to it last week when I was trying again with everything up
to date. Confirmed that I'm using Kinect SDK 2.0.
[image: image.png]

SirLynix, 2020-06-01:
Okay, but did you try SDK 2 examples? Like Coordinate mapping?

Also, what is the CPU/GPU usage when running obs-kinect?

SirLynix, 2020-10-21:
Closed for no reply

closed 2020-10-21 by SirLynix