SirLynix/obs-kinect #20: Body & depth glitches sometimes?

marv3lls, 2020-05-28:
I've gotten it where just depth will render my close body pretty well but Body & Depth sometimes puts a body up and to the left of where I am? I'd love to fork and help ya play with this stuff, it's just been a long time since I developed for Windows more than PHP lmao.
[image]

marv3lls, 2020-05-28:
Ah, also important, it only turns on when I set depth delay frames to one.

SirLynix, 2020-05-29:
This seems to be a bug with the CPU depth mapping. Could you confirm it also happen when unchecking "GPU depth-mapping"? I'm looking at it.

SirLynix, 2020-05-29:
Found and fixed, the fix will be available in 0.3 RC3 (which should be the last one before 0.3 release).
Thank you for the report.

commit 5eadce8c (2020-05-29): Fix body CPU mapping (fix #20)

+ allow dirty depth option with body
closed 2020-05-29 by SirLynix