SirLynix/obs-kinect #21: Kinect v1 servo-support

oneohtrix, 2020-06-05:
Kinect v1 (and v2 as well?) has a servo, which can control the vertical orientation.

SirLynix, 2020-06-05:
Have you tried obs-kinect 0.3 (RC2)? It has support for that (check the [releases](https://github.com/SirLynix/obs-kinect/releases)).

Kinect v2 doesn't have servo unfortunately.

oneohtrix, 2020-06-08:
Hi, obs-kinect 0.3 (RC2) seems to work, but it has a few quirks.

For some reason, the Kinect device gets displayed in the list, however, choosing it doesn't actually send any commands to Kinect. 

If I unplug and plug it backand choose it from the list before it gets it's USB device path appended to it, it seems to work. 

I'm guessing there might be a race condition somewhere (I'm not aware of any other software that'd try to use the Kinect besides the Kinect Developer Toolkit browser, which wasn't launched at the time)

Anyway: When I change scenes in OBS, the camera elevation always seems to reset to 0°.
I'd love to have different elevations set for different scenes, in order to switch between them (e.g. one pointing at my face, and then switch to keyboard view)

SirLynix, 2020-06-08:
> For some reason, the Kinect device gets displayed in the list, however, choosing it doesn't actually send any commands to Kinect.
> 
> If I unplug and plug it back in, and then choose it from the list before it gets it's USB device path appended to it, it seems to work.
> 
> I'm guessing there might be a race condition somewhere (I'm not aware of any other software that'd try to use the Kinect besides the Kinect Developer Toolkit browser, which wasn't launched at the time)

The USB device path is just part of the name returned by the Windows SDK, and is only fetched when refreshing devices.

There should be an error in your OBS logs explaining why obs-kinect can't access the kinect, could you copy and paste it?

> Anyway: When I change scenes in OBS, the camera elevation always seems to reset to 0°.
> I'd love to have different elevations set for different scenes, in order to switch between them (e.g. one pointing at my face, and then switch to keyboard view)

Do you have a source per scene? If yes then it seems like a bug. I'll check it asap.

SirLynix, 2020-06-08:
I just checked the Kinect elevation per scene and it works for me. But you have to check the "disable if invisible" on both sources. If multiple Kinect v1 sources are active at the same time with a different camera elevation it will take one of them in a random way.

SirLynix, 2020-10-21:
Closed for no reply

closed 2020-10-21 by SirLynix