SirLynix/obs-kinect #23: Nothing shown up when added KinectSource

haxpor, 2020-06-23:
Hi, thanks for this project. Looks promising.

I tested on Windows 10 build 19041.264 with Obs version 25.0.8 64-bit. Properly installed all dependencies as said.

Downloaded and tested both version 3 RC 1 and 2, both didn't show video feedback after adding KinectSource despite waiting for a long while.

[image]

SirLynix, 2020-06-23:
Hello, could you check if OBS logs say something about this? Search for "obs-kinect".

haxpor, 2020-06-23:
Ok, I found the following.
I use kinect v1

```
06:01:37.143: User added source 'KinectSource' (kinect_source) to scene 'Scene'
06:01:37.239: [obs-kinect] failed to initialize Kinect: Not supported
```

SirLynix, 2020-06-24:
This is weird. I'm not aware of Kinect v1 models that aren't supported.

Could you try to install [Kinect for Windows Developer Toolkit](https://www.microsoft.com/en-us/download/details.aspx?id=40276) and run some examples? (like Color Basics-D2D, Depth Basics-D2D and such) Those are officials examples from Microsoft, I suppose they should handle this better than I do.

haxpor, 2020-06-24:
Thank you for suggestion!
Tested examples in toolkits, works just fine. Then back to test in OBS, it works now!

Not sure what happens. But it works after I installed **additionally** the following twos (possibly only first would suffice)

* [Kinect for Windows SDK 1.8](https://www.microsoft.com/en-us/download/confirmation.aspx?id=40278)
* [Kinect for Windows Developer Toolkit v1.8 ](https://www.microsoft.com/en-us/download/details.aspx?id=40276)

closed 2020-06-24 by haxpor

SirLynix, 2020-06-24:
Glad it worked for you. But this is weird af, I will add a note to the readme about this (it seems Kinect for Windows SDK 1.8 has a bigger role than I though).

smellyfinger, 2020-09-08:
Just a heads up you blurred the Kinect device but right under it says the device name again.