SirLynix/obs-kinect #25: any news or workaround to use the addon into Streamlabs?

moebiussurfing, 2020-07-03:
Sorry to ask, because you mention on readme that is not compatible...

I copied the plugin files into:
C:\Program Files\Streamlabs OBS\resources\app.asar.unpacked\node_modules\obs-studio-node\obs-plugins
C:\Program Files\Streamlabs OBS\resources\app.asar.unpacked\node_modules\obs-studio-node\data\obs-plugins\obs-kinect

But the plugin is not detected as a widget.

It works fine in Obs Studio.  Thanks
There’s some workaround?

SirLynix, 2020-07-04:
Hello, unfortunately as far as I know SLOBS doesn't support thirdparty plugins (I would love to be proved wrong about this though).

Do you know some SLOBS plugins (or better yet, OBS Studio plugins that work with SLOBS)? I would love to add support for SLOBS but I just don't know how.

moebiussurfing, 2020-07-04:
Thanks for your reply. 

I read around the forum that there are some compatible plugins:
https://howto.streamlabs.com/streamlabs-obs-9/list-of-compatible-plugins-8404?postid=28232#post28232
but I am so newbie and never used SLOBS before.

It looks that is still closed. Suggestions only...
https://howto.streamlabs.com/streamlabs-obs-9/plugin-from-obs-studio-to-streamlab-3780

closed 2020-07-04 by moebiussurfing