SirLynix/obs-kinect #26: VCRUNTIME140.DLL missing

Liskowskyy, 2020-07-11:
I was trying to run this plugin on my OBS Studio 25.0.8 64-bit, but it outputted `22:01:31.677: LoadLibrary failed for '../../obs-plugins/64bit/obs-kinect.dll': The specified module could not be found.
22:01:31.677:  (126)
22:01:31.677: Module '../../obs-plugins/64bit/obs-kinect.dll' not loaded` to the log.
I ran Dependency Walker just like you said in one of the other issues, and it displayed some errors regarding DLLs.
Here's the Dependency Walker output: 
[obs-kinect.zip](https://github.com/SirLynix/obs-kinect/files/4907539/obs-kinect.zip)

I've already installed the Visual Studio redistributables you mentioned (both 32 and 64-bit)

Liskowskyy, 2020-07-11:
Closing this issues, the problem is solved by installing 0.3 RC2 instead of 0.2

closed 2020-07-11 by Liskowskyy

SirLynix, 2020-07-12:
It seems like you want support for the Kinect v1/for 360, so yes 0.3 is your only option.

It's about time that I fully release 0.3.