SirLynix/obs-kinect #28: OBS crashes when using KinectSDK background removal

pop13-13, 2020-07-12:
Using the KinectSDK background removal option crashes OBS, crash log below:

`Unhandled exception: c0000005
Date/Time: 2020-07-12, 16:33:25
Fault address: 7FEC6835553 (c:\program files\obs-studio\obs-plugins\64bit\obs-kinect-sdk10.dll)
libobs version: 25.0.8 (64-bit)
Windows version: 6.1 build 7601 (revision: 24384; 64-bit)
CPU: Intel(R) Core(TM) i7-7700K CPU @ 4.20GHz

Thread 29B4 (Crashed)
Stack            EIP              Arg0             Arg1             Arg2             Arg3             Address
000000000BE5EFB0 000007FEC6835553 0000000000000000 0000000000000000 0000000007E20950 0000000007E20A40 obs-kinect-sdk10.dll!KinectSdk10Device::ThreadFunc+0x6b3
000000000BE5FDA0 000007FEB4FE31BE 0000000000000000 0000000000000000 0000000000000000 0000000007E20A40 obs-kinect.dll!std::thread::_Invoke<std::tuple<<lambda_8a31a3b216a40e3f887a456a81f41ba2> >,0>+0x1e
000000000BE5FDD0 000007FEF966BEED 0000000000000000 0000000000000000 0000000000000000 0000000000000000 ucrtbase.dll!0x7fef966beed
000000000BE5FE00 00000000770E570D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 kernel32.dll!0x770e570d
000000000BE5FE30 000000007724385D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 ntdll.dll!0x7724385d

Thread 1EF8
Stack            EIP              Arg0             Arg1             Arg2             Arg3             Address
00000000002FBC98 0000000077259E3A 0000000000000001 0000000000000001 00000000002FBCE8 000007FEFAEF96DF ntdll.dll!0x77259e3a
00000000002FBCA0 000007FEFD1E1430 00000000002FBDD8 00000000002FBDD0 0000000000000000 0000000000000000 kernelbase.dll!0x7fefd1e1430
00000000002FBDA0 00000000770F14A3 00000000002FBE60 00000000FFFFFFFF 0000000000000000 0000000000000020 kernel32.dll!0x770f14a3
00000000002FBE30 0000000076FE8F8D 0000000000000000 00000000002FC010 00000000005A6201 0000000000000024 user32.dll!0x76fe8f8d
00000000002FBED0 0000000076FE6272 00000000005AA9C0 0000000000000003 0000000000000000 00000000005A6298 user32.dll!0x76fe6272
00000000002FBF10 000007FEB76A7C01 000000013F6C5800 0000000000000000 0000000000594200 00000000080B0200 qt5core.dll!0x7feb76a7c01
00000000002FF090 000007FEB5A7AA69 0000000000000000 0000000000000014 0000000000570030 0000000000570030 qwindows.dll!0x7feb5a7aa69
00000000002FF0C0 000007FEB7657442 000000000056CAB0 0000000000000000 00000000002FF640 000007FEB7757F98 qt5core.dll!0x7feb7657442
00000000002FF140 000007FEB765AB3A 000000013F775A63 00000000002FF580 00000000002FF640 000000000058F1A0 qt5core.dll!0x7feb765ab3a
00000000002FF1B0 000000013F51654A 0000000000000000 000000000058E8E0 0000000000000000 000000000057EC70 obs64.exe!run_program+0x7ca
00000000002FF540 000000013F5186F1 0000000000570030 0000000000000000 0000000000000000 0000000000000000 obs64.exe!main+0x671
00000000002FF710 000000013F68FDC4 0000000000000001 0000000000000000 0000000000000000 0000000000000000 obs64.exe!WinMain+0x154
00000000002FF7A0 000000013F68F47E 0000000000000000 0000000000000000 0000000000000000 0000000000000000 obs64.exe!__scrt_common_main_seh+0x106
00000000002FF7E0 00000000770E570D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 kernel32.dll!0x770e570d
00000000002FF810 000000007724385D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 ntdll.dll!0x7724385d

Thread B04
Stack            EIP              Arg0             Arg1             Arg2             Arg3             Address
00000000023FF968 0000000077259E3A 000000000000001B 0000000000000000 00000000771FAA20 00000000771FAA20 ntdll.dll!0x77259e3a
00000000023FF970 000000007720332D 0000000000000000 0000000000000000 0000000000000000 000000361BF94921 ntdll.dll!0x7720332d
00000000023FFC10 00000000770E570D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 kernel32.dll!0x770e570d
00000000023FFC40 000000007724385D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 ntdll.dll!0x7724385d

Thread 34E8
Stack            EIP              Arg0             Arg1             Arg2             Arg3             Address
0000000003E5FCC8 000000007725B18A 0000000000040000 0000000000000000 0000000000592080 0000000000571DF0 ntdll.dll!0x7725b18a
0000000003E5FCD0 00000000772E96E4 0000000000000008 0000000000010007 0000000000571DF0 0000000000593D00 ntdll.dll!0x772e96e4
0000000003E5FF60 00000000770E570D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 kernel32.dll!0x770e570d
0000000003E5FF90 000000007724385D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 ntdll.dll!0x7724385d

Thread 2C5C
Stack            EIP              Arg0             Arg1             Arg2             Arg3             Address
000000000404F738 000000007725B18A 0000000000060000 0000000000000000 0000000007DECEB0 0000000000571DF0 ntdll.dll!0x7725b18a
000000000404F740 00000000772E96E4 0000000000000008 0000000000010006 0000000000571DF0 0000000007FCC850 ntdll.dll!0x772e96e4
000000000404F9D0 00000000770E570D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 kernel32.dll!0x770e570d
000000000404FA00 000000007724385D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 ntdll.dll!0x7724385d

Thread 2260
Stack            EIP              Arg0             Arg1             Arg2             Arg3             Address
000000000289F6B8 00000000772598CA 0000000000000000 0000000003212DC0 0000000000000000 0000000000000000 ntdll.dll!0x772598ca
000000000289F6C0 000007FEFD1E10AC 0000000000000000 0000000000000000 0000000000000000 0000000000000354 kernelbase.dll!0x7fefd1e10ac
000000000289F760 000007FEF2E3DC4F 0000000000000000 0000000000000000 0000000000000000 00000000000003B6 obs.dll!os_event_timedwait+0xf
000000000289F790 000007FEF2E68589 00000000031B88C0 0000000000000000 000000000316B8E0 0000000000000000 obs.dll!obs_hotkey_thread+0x1b9
000000000289F7F0 000007FEFA4358EA 0000000000000000 0000000000000000 0000000000000000 000000000316B8E0 w32-pthreads.dll!ptw32_threadStart+0x8a
000000000289F880 000007FEF966BEED 0000000000000000 0000000000000000 0000000000000000 0000000000000000 ucrtbase.dll!0x7fef966beed
000000000289F8B0 00000000770E570D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 kernel32.dll!0x770e570d
000000000289F8E0 000000007724385D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 ntdll.dll!0x7724385d

Thread 28C8
Stack            EIP              Arg0             Arg1             Arg2             Arg3             Address
0000000005A0F758 00000000772598CA 0000000000000048 000007FE00000001 0000000000000000 0000000000000000 ntdll.dll!0x772598ca
[... 666 lines cut ...]
000007FEB4F10000-000007FEB4F26000 C:\Program Files\obs-studio\obs-plugins\64bit\obs-virtualoutput.dll
000007FEB4EF0000-000007FEB4F01000 C:\Program Files\obs-studio\obs-plugins\64bit\obs-vst.dll
000007FECA320000-000007FECA32B000 C:\Program Files\obs-studio\obs-plugins\64bit\obs-x264.dll
000007FEB4ED0000-000007FEB4EE1000 C:\Program Files\obs-studio\obs-plugins\64bit\rtmp-services.dll
000007FEAC970000-000007FEACA03000 C:\Program Files\obs-studio\obs-plugins\64bit\text-freetype2.dll
000007FEFBE00000-000007FEFBE07000 C:\Windows\System32\wshtcpip.dll
000007FEFC7C0000-000007FEFC81B000 C:\Windows\system32\DNSAPI.dll
000007FEC39A0000-000007FEC39AC000 C:\Program Files\obs-studio\obs-plugins\64bit\vlc-video.dll
000007FEB4E90000-000007FEB4EC4000 C:\Program Files\VideoLAN\VLC\libvlc.dll
000007FEAB4F0000-000007FEAB7A4000 C:\Program Files\VideoLAN\VLC\libvlccore.dll
000007FEF6B40000-000007FEF6B48000 C:\Windows\system32\rasadhlp.dll
000007FEACED0000-000007FEACEE6000 C:\Program Files\obs-studio\obs-plugins\64bit\win-capture.dll
000007FEF8C50000-000007FEF8CA3000 C:\Windows\System32\fwpuclnt.dll
000007FEACB00000-000007FEACB13000 C:\Program Files\obs-studio\obs-plugins\64bit\win-decklink.dll
000007FEACAD0000-000007FEACAFF000 C:\Program Files\obs-studio\obs-plugins\64bit\win-dshow.dll
000007FEACAB0000-000007FEACAC6000 C:\Windows\system32\devenum.dll
000007FEFC3D0000-000007FEFC3FD000 C:\Windows\system32\ntmarta.dll
000007FEFA6C0000-000007FEFA6CD000 C:\Windows\system32\msdmo.dll
000007FEB9700000-000007FEB9707000 C:\Program Files\obs-studio\obs-plugins\64bit\win-mf.dll
000007FEB8DB0000-000007FEB8DBD000 C:\Program Files\obs-studio\obs-plugins\64bit\win-wasapi.dll
000007FEFC5F0000-000007FEFC5FA000 C:\Windows\system32\credssp.dll
000007FEFC730000-000007FEFC788000 C:\Windows\system32\schannel.DLL
000007FEFC440000-000007FEFC45B000 C:\Windows\system32\GPAPI.dll
000007FEB8F40000-000007FEB8F49000 C:\Windows\system32\SensApi.dll
`

SirLynix, 2020-07-14:
This is a weird one. I had it myself a few weeks ago but it never showed up again. Is it happening everytime you try to use the KinectSDK background? Could you try to put color, color with body greenscreen and color with depth greenscreen sources at the same time? (it will requires the same info from the Kinect which are used by the KinectSDK background removal).

pop13-13, 2020-07-14:
I will try that. Everything works (colour, depth, IR and the other greenscreen methods). I once managed to get it working without a crash, but I was unable to reproduce that.The SDK example of the background removal works on my PC, but the OBS plugin doesen't.And just saying that this is with a Kinect for Xbox 360.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/SirLynix/obs-kinect/issues/28#issuecomment-658061094",
"url": "https://github.com/SirLynix/obs-kinect/issues/28#issuecomment-658061094",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

SirLynix, 2020-10-21:
Hi!

Could you give [this version](https://github.com/SirLynix/obs-kinect/issues/33#issuecomment-713488630) a try? I know it's a release for Azure Kinect but it also includes some fixes since obs-kinect 0.3 RC2.

pop13-13, 2020-10-21:
Thanks, will look into it

SirLynix, 2020-11-12:
Hello, did you try the version I sent you? Does it work?

pop13-13, 2020-11-12:
Shit, totally forgot about it, sorry. Will look into it hopefully this weekend

pop13-13, 2020-11-17:
Just tried it, same issue, crashes if I select KinectSDK background removal.Attached the crash logs

Date/Time: 2020-11-17, 19:42:09
Fault address: 7FEA0852E13 (c:\program files\obs-studio\obs-plugins\64bit\obs-kinect-sdk10.dll)
libobs version: 26.0.2 (64-bit)
Windows version: 6.1 build 7601 (release: unavailable; revision: 24384; 64-bit)

Thread 19A8 (Crashed)
000000000DABEA50 000007FEA0852E13 000000000DABEB88 000000013F334BB5 000000000DABEF40 000000000DABEFA0 obs-kinect-sdk10.dll!<lambda_9aeb2d1c84cf870cc81528311012b40a>::operator()+0xa3
000000000DABEB60 000007FEA0854BFB 0000000000000000 000007FEA0859D40 0000000000000000 0000000000001AE0 obs-kinect-sdk10.dll!KinectSdk10Device::RetrieveDepthFrame+0x9b
000000000DABECD0 000007FEA0855B2A 0000000000000000 0000000000000000 0000000007C3B490 0000000007C3B4C0 obs-kinect-sdk10.dll!KinectSdk10Device::ThreadFunc+0x98a
000000000DABFAD0 000007FEA08332FE 0000000000000000 0000000000000000 0000000000000000 0000000007C3B4C0 obs-kinect.dll!std::thread::_Invoke<std::tuple<<lambda_8a31a3b216a40e3f887a456a81f41ba2> >,0>+0x1e
000000000DABFB00 000007FEB372BEED 0000000000000000 0000000000000000 0000000000000000 0000000000000000 ucrtbase.dll!0x7feb372beed
000000000DABFB30 00000000773A570D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 kernel32.dll!0x773a570d
000000000DABFB60 000000007760385D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 ntdll.dll!0x7760385d

Thread 1768
000000000016C038 0000000077619E3A 0000000000000001 0000000000000001 000000000016C088 000007FEF97B96DF ntdll.dll!0x77619e3a
000000000016C040 000007FEFD6E1430 0000000000000000 0000000000000000 0000000000000001 0000000000000000 kernelbase.dll!0x7fefd6e1430
000000000016C140 00000000773B14A3 000000000016C200 00000000FFFFFFFF 0000000000000000 0000000000000020 kernel32.dll!0x773b14a3
000000000016C1D0 00000000774C8F8D 0000000000000000 000000000016C3B0 00000000002C8601 0000000000000024 user32.dll!0x774c8f8d
000000000016C270 00000000774C6272 00000000002C7E20 0000000000000003 0000000000000000 00000000002C8678 user32.dll!0x774c6272
000000000016C2B0 000007FEABFE7C01 000000013F4FA200 0000000000000000 00000000002B4C00 000000000889F000 qt5core.dll!0x7feabfe7c01
000000000016F430 000007FEA9E4AA69 0000000000000000 0000000000000014 00000000002900B0 00000000002900B0 qwindows.dll!0x7fea9e4aa69
000000000016F460 000007FEABF97442 000000000028CB30 0000000000000000 000000000016F9E0 000007FEAC097F98 qt5core.dll!0x7feabf97442
000000000016F4E0 000007FEABF9AB3A 000000013F5B3D74 000000000016F920 000000000016F9E0 00000000002AF640 qt5core.dll!0x7feabf9ab3a
000000000016F550 000000013F33772A 0000000000000000 00000000002AEA30 0000000000000000 000000000029EE10 obs64.exe!run_program+0x7ca
000000000016F8E0 000000013F339855 00000000002900B0 0000000000000000 0000000000000000 0000000000000000 obs64.exe!main+0x5e5
000000000016FAB0 000000013F4C29B4 0000000000000001 0000000000000000 0000000000000000 0000000000000000 obs64.exe!WinMain+0x154
000000000016FB40 000000013F4C21EA 0000000000000000 0000000000000000 0000000000000000 0000000000000000 obs64.exe!__scrt_common_main_seh+0x106
000000000016FB80 00000000773A570D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 kernel32.dll!0x773a570d
000000000016FBB0 000000007760385D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 ntdll.dll!0x7760385d

Thread 176C
000000000237FA58 0000000077619E3A 000000000000001B 0000000000000000 00000000775BAA20 00000000775BAA20 ntdll.dll!0x77619e3a
000000000237FA60 00000000775C332D 0000000000000000 0000000000000000 0000000000000000 0000000038BD5C30 ntdll.dll!0x775c332d
000000000237FD00 00000000773A570D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 kernel32.dll!0x773a570d
000000000237FD30 000000007760385D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 ntdll.dll!0x7760385d

Thread 145C
00000000029EFA78 000000007761B18A 0000000000000000 0000000000000000 00000000002B2210 0000000000291E70 ntdll.dll!0x7761b18a
00000000029EFA80 00000000776A96E4 0000000000000008 0000000000010007 0000000000291E70 00000000002B3E90 ntdll.dll!0x776a96e4
00000000029EFD10 00000000773A570D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 kernel32.dll!0x773a570d
00000000029EFD40 000000007760385D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 ntdll.dll!0x7760385d

Thread 1464
000000000403F578 000000007761B18A 0000000000000000 0000000000000000 000000001AA486F0 0000000000291E70 ntdll.dll!0x7761b18a
000000000403F580 00000000776A96E4 0000000000000008 0000000000010006 0000000000291E70 000000000875EA20 ntdll.dll!0x776a96e4
000000000403F810 00000000773A570D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 kernel32.dll!0x773a570d
000000000403F840 000000007760385D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 ntdll.dll!0x7760385d

Thread 1468
000000000415FAE8 00000000776198CA 0000000000000000 0000000003214840 0000000000000000 0000000000000000 ntdll.dll!0x776198ca
000000000415FAF0 000007FEFD6E10AC 0000000000000000 0000000000000000 0000000000000000 0000000000000354 kernelbase.dll!0x7fefd6e10ac
000000000415FB90 000007FEABCEDCDF 0000000000000000 0000000000000000 0000000000000000 00000000000003B6 obs.dll!os_event_timedwait+0xf
000000000415FBC0 000007FEABD184B9 0000000003178E40 0000000000000000 0000000003164260 0000000000000000 obs.dll!obs_hotkey_thread+0x1b9
[... cut ...]
000000001D44FBB8 0000000077619E3A 000000001AC49DC0 0000000000000070 000007FEF9100400 000000001AC49DC0 ntdll.dll!0x77619e3a
000000001D44FBC0 000007FEF90DA8AF 0000000000000000 0000000000000001 0000000000000000 0000000000000000 winmm.dll!0x7fef90da8af
000000001D44FD10 00000000773A570D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 kernel32.dll!0x773a570d
000000001D44FD40 000000007760385D 0000000000000000 0000000000000000 0000000000000000 0000000000000000 ntdll.dll!0x7760385d

Thread F7C
Stack

PigeonRubbishStudios, 2020-11-24:
Having the same issue. Can post crash log if necessary. Let me know!

PigeonRubbishStudios, 2020-11-24:
Ok let me change my previous statement: here's my crash log.

However, I did find a semi fix (maybe). When I select Background Removal I had a few seconds before the app crashed. In that few seconds I was able to hit ok in the Kinect source properties to apply those changes I had made. OBS proceeded to crash, but, when I opened OBS back up, the background removal had been applied and I can no longer replicate the crash even if I make a new scene and add a new Kinect Source. Still is a weird issue but hopefully this works to "fix" your issue.

[Crash 2020-11-24 10-02-59.txt](https://github.com/SirLynix/obs-kinect/files/5591252/Crash.2020-11-24.10-02-59.txt)

SirLynix, 2020-11-24:
It's quite interesting to know that, thanks for the info. I'll try to work on it asap (but i'm quite busy this week).

zxcvbs, 2022-08-01:
Same problem, fixed applying changes before crash

SebastienGllmt, 2022-09-19:
I seem to be having a similar issue

I looked at the logs and it said
```
[obs-kinect] [azure] SDK warning: is_fw_version_compatable(). Firmware version for RGB is 1.6.102. Consider upgrading to 1.6.110 or newer. (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\depth\depth.c:88)

[obs-kinect] [azure] SDK warning: is_fw_version_compatable(). Firmware version for Depth is 1.6.75. Consider upgrading to 1.6.79 or newer. (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\depth\depth.c:88)
```

so I followed [the instructions](https://learn.microsoft.com/en-us/azure/kinect-dk/update-device-firmware) for upgrading my firmware.

However, now I get a new error

```
[obs-kinect] [azure] SDK error: dynlib_create(). Failed to load DLL depthengine_2_0 with error code: 126 (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\dynlib\dynlib_windows.c:115)

[obs-kinect] [azure] SDK error: deloader_init_once(). Failed to Load Depth Engine Plugin (depthengine). Depth functionality will not work (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\deloader\deloader.cpp:75)

[obs-kinect] [azure] SDK error: deloader_init_once(). Make sure the depth engine plugin is in your loaders path (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\deloader\deloader.cpp:76)

[obs-kinect] [azure] SDK error: deloader_depth_engine_create_and_initialize(). Failed to load depth engine plugin (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\deloader\deloader.cpp:113)

[obs-kinect] [azure] SDK error: depth_engine_start_helper(). Depth engine create and initialize failed with error code: 108. (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\dewrapper\dewrapper.c:154)

[obs-kinect] [azure] SDK error: deresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in depth_engine_start_helper() (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\dewrapper\dewrapper.c:160)

[obs-kinect] [azure] SDK error: depth_engine_start_helper(dewrapper, dewrapper->fps, dewrapper->depth_mode, &depth_engine_max_compute_time_ms, &depth_engine_output_buffer_size) returned failure in depth_engine_thread() (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\dewrapper\dewrapper.c:194)

[obs-kinect] [azure] SDK warning: capturesync_add_capture(). Capture Error Detected, Depth  (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\capturesync\capturesync.c:200)

[obs-kinect] [azure] SDK error: dewrapper_start(). Depth Engine thread failed to start (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\dewrapper\dewrapper.c:552)

[obs-kinect] [azure] SDK error: dewrapper_start(depth->dewrapper, config, depth->calibration_memory, depth->calibration_memory_size) returned failure in depth_start() (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\depth\depth.c:398)

[obs-kinect] [azure] SDK error: cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming() (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\depth_mcu\depth_mcu.c:359)

[obs-kinect] [azure] SDK error: depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063 (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\depth_mcu\depth_mcu.c:362)

[obs-kinect] [azure] SDK error: depth_start(device->depth, config) returned failure in k4a_device_start_cameras() (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\sdk\k4a.c:895)

[obs-kinect] [azure] Failed to start cameras!
```

Maybe this error isn't surprising because `1.0rc1` of this library includes all the dlls of the Azure Kinect SDK such as `/bin/64bit/k4a.dll`, but does not include `depthengine_2_0.dll` (despite the fact depthengine is part of `Azure Kinect SDK v1.4.1\sdk\windows-desktop\x86\release\bin`

Perhaps this DLL needs to be added to https://github.com/SirLynix/obs-kinect/tree/v1.0-rc1/xmake-repo/packages/k ?

SirLynix, 2022-09-19:
Hi!

The depth engine DLL isn't distributed as part of the Azure Kinect SDK, if I recall correctly this is because it cannot be redistributed by a thirdparty.

This is the reason you have to install the Azure Kinect SDK to use this plugin.

As for why it doesn't find the depth engine since installing it, perhaps it's a version mismatch? Try to replace the sdk dll shipped with the plugin by the ones from your Azure Kinect SDK installation.

sebisapussy, 2024-12-21:
was this ever fixed?? cause I'm having the same issue when I enabled KinectSDK background removal it will either just straight up not work and display nothing on the screen or it displays a crash message the same as the one above. I've tried switching versions without any success and tried to build the master hoping the issue might of been fixed but i cant build cause it shows this error "'obs-module.h': No such file or directory", not to mention I've tried moving the KinectBackgroundRemoval180_64.dll file to different locations without any success and i don't think its a driver thing cause the background removal from Xbox Kinect toolkit v1.8 works fine though so idk

sebisapussy, 2024-12-22:
edit: i managed to fix it and get obs working with kinect background removal when I used obs version 27.1.3 with the plugin obs-kinect-0.3-rc2