SirLynix/obs-kinect #29: Offset?

ShocWave, 2020-08-05:
I tried playing around with this today, and I noticed that unless I stand 1-2 meters away from the Kinect, there seems to be a very huge offset between the color picture and the depth information. Every seems to be shifted over by 10-20 pixels, leaving some kind of strange looking shadow.

I assume this is because the two cameras are quite far apart from each other, so some distance is needed for it to combine everything correctly.

Is there a possibility that in the future, you could add some manual controls to align the depth and color images?
I'm not able to place the camera far away from me due to space constraints.

SirLynix, 2020-08-05:
Hello,

Could you please tell me what version of obs-kinect you're using, along with the source setup you have.

Why do you need to combine color and depth streams? The greenscreen effect has depth to color conversion and will handle that for you.

Could you please post an image of what's happening and describe how you're using obs-kinect sources?

Thank you.

ShocWave, 2020-08-06:
Sorry, maybe I wasn't very good at explaining it. I've attached an image here: https://imgur.com/y45rT7O
Take a look at my hand and thumb, and the side of my head, it seems like the depth sensor thinks my body is shifted to the left.

The further back I move, the less I notice this.

SirLynix, 2020-08-06:
What you are experiencing isn't caused by a bug / offset, it's an **occlusion shadow**.
[image]

This is caused by the color and IR camera being one centimer apart (+ the IR emitter and IR camera being one centimer apart too).
The only way to counteract this is by having a better filter, I've been experimenting with this but I'm lacking some skills.

Could you try using "Body" instead of "Depth" as filter type? This will use Microsoft body detection instead of raw depth values which may improve the result.

ShocWave, 2020-08-06:
Yeah, I'm aware this is not really a bug, and caused by the sensors being apart from each other.
Would it be possible to include some manual adjustments to shift the IR and Color data so that it can be adjusted to work with closer distances?

SirLynix, 2020-08-07:
Well, yes it's possible to let the user offset the depth image but I really don't think it will help. As I said it's not an offset problem but a missing information problem. If you offset the depth image to the left side you're gonna be cropped on the right one.

SirLynix, 2020-10-21:
Closed for no reply. Unfortunately there's not much I can do here.

closed 2020-10-21 by SirLynix