SirLynix/obs-kinect #31: Minimum depth

Shiaoran, 2020-09-14:
Hi, when getting too close to the Kinect when using the faux green screen, the image isn't captured. Is this a hardware limitation of the 3D scan?
If not, could you make it possible for it to detect objects closer than the current minimum?
If it is, could you make the 2D camera replace the missing image only in the spot that got too close?

I've been messing with using the 2 sources, but unless I can find a way to have an image mask that follows me, it's defeating the purpose of the green screen.

SirLynix, 2020-09-14:
Hi.

Unfortunately this is a hardware limitation, and I can't just detect close pixels.
Technically speaking, the Kinect computes a depth value in millimeters for every pixels, when it can't compute the depth for a pixel it gives it a special value, which is processed by obs-kinect as "nothing" when using greenscreen.

Problem is, it's not just when something is too close I get this value, I can get it if the pixel is too far away, or if it's depth cannot be processed (see "transparency shadow" in the README) or is fluctuating (depth values around objects can be available only some frames and not others).

Check the depth stream, you'll see fluctuating black pixels in the background and around objects, and on objects that are too close. This is the problem: I cannot tell apart those cases, as it's reported the same by the Kinect.

So yeah sorry, I can't do much against this. The closer depth value which can be read by the Kinect depends on its model (80cm for the Kinect v1, 40cm for the Kinect for Windows and 50cm for the Kinect v2).

Shiaoran, 2020-09-14:
I suspeced it so. But is my suggestion for filling in the gaps with 2D feasible?

SirLynix, 2020-09-14:
Unfortunately not, because I can't detect if it's a gap because it's closer or for some other reason. if I were to give you a way to fill the gaps with an image it'll just look ugly

Shiaoran, 2020-09-14:
Too bad, thanks for explaining.

closed 2020-09-16 by SirLynix