SirLynix/obs-kinect #38: Upside down body detection

MarcoWilli, 2020-10-30:
Hi!

First: Thanks to your great piece of software I´m able to use my Kinect v2 for my streaming setup (on Windows at least so far) - so, cool stuff and thanks!

Due to the way I mounted my Kinect (upside down) body detection doest not work. I did not think about that when I chose my mounting setup point because flipping the image is just a click. I guess you can not do anything about that behaviour?!

I´m strongly looking forward to Linux support aswell!!

Kind regards =)

PS: You do mention in the 0.3 notes that German translation is not fully done but I guess this is just a false reporting?! If not, I can do translations.

SirLynix, 2020-10-30:
Hi!

Well unfortunately (as confirmed by Microsoft [here](https://social.msdn.microsoft.com/Forums/en-US/d45804d3-ca0a-4459-9781-bfb087eea5f0/kinect-in-inverted-position?forum=kinectsdk) and [there](https://social.msdn.microsoft.com/Forums/en-US/5673678c-3835-4eab-afc6-e9dbb846f7fa/installing-kinect-upside-down-mounted-on-the-ceiling?forum=kinectsdk)) there's nothing I can do about this, Kinect v1/v2 body tracking is only working with the camera is upright (Azure Kinect does have a setting for that though).

So yeah, there shouldn't be any problem with color / depth / infrared streams but I'm afraid you won't have body detection without putting the camera upright.

About german translation, it's complete for now but I will add some translations keys until the next release, so I could need some help about is soon!

MarcoWilli, 2020-10-31:
Ah ok. I did not look it up online though this is the first step I do usually :p

As I'm watching this project I guess I will get anounced if there's something new relevant to translate and I'll have a look at it.
You may also contact me for the Linux thing if that can give you some help. ATM I am able to use the microphone (which I don't need ^^) and get output of the camera by using the bundle (pyhon?) app "protonect" that ships with libfreenect2. But that's OT anyway... =)

closed 2020-12-26 by SirLynix