SirLynix/obs-kinect #39: Kinect source in OBS

Jenko1, 2020-11-11:
Hi All, I have downloaded all files to use my Kinect as webcam/green screen in obs. i have copied the files over to obs-studio on my C drive, unfortunately when I open OBS their is no Kinect source when I goto add a new source.

Worked on my other laptop but unfortunately the graphic's card isn't good enough hence why I tried on the new laptop.

Any ideas would be most apricated.

SirLynix, 2020-11-11:
Did you install [Visual Studio 2019 redistributables](https://aka.ms/vs/16/release/vc_redist.x64.exe)? See the FAQ : [I copied the files and the source doesn't show up](https://github.com/SirLynix/obs-kinect#i-copied-the-files-and-the-source-doesnt-show-up).

Jenko1, 2020-11-11:
Thank you for your reply. Download and all now working great. Thanks again.

closed 2020-11-11 by Jenko1