SirLynix/obs-kinect #40: Kinect

Jenko1, 2020-12-05:
Evening all, So had the Xbox Kinect working fine for weeks now, come to use today and the kinect comes on for around 10 seconds then goes off an then comes back on again. Just keeps doing the same, image is viable when kinect is on but then freezes when it goes off.

Any ideas what the cause of this maybe, i would really apricate any assistance.

SirLynix, 2020-12-06:
Hi!

I'm guessing you have a Kinect v2 (for xbox one), this is unfortunately a known hardware issue. As far as I know it's the cooling fan that's not starting.
I have the same issue, in my case it seems the fan finally starts after about 10-20min of rebooting (maybe more some times) and gets stable from there.

However it seems it's relatively easy to fix, see theses links:
https://www.reddit.com/r/xboxone/comments/4vooz5/psa_if_your_kinect_is_shutting_off_randomly_this/
https://forums.anandtech.com/threads/what-is-best-fix-for-xbox-kinect-fan-failure.2546859/
https://www.youtube.com/watch?v=BoRK3jJVMLM

Jenko1, 2020-12-06:
Hi,
Thanks for this, problem is a soon as you activate the Kinect and get notifications coming up see attached. If I’m honest I’ve never heard the fan activate on the Kinect.[Image.jpeg][Image.jpeg]

https://www.reddit.com/r/xboxone/comments/4vooz5/psa_if_your_kinect_is_shutting_off_randomly_this/
https://forums.anandtech.com/threads/what-is-best-fix-for-xbox-kinect-fan-failure.2546859/
https://www.youtube.com/watch?v=BoRK3jJVMLM

SirLynix, 2020-12-06:
Oh you mean you have USB plug in/out notification? This may indicate the Kinect isn't plugged in a USB 3 port (or maybe the port has issues).

Just to be sure this problem isn't related to obs-plugin, this issue happens also with the official SDK examples?

Jenko1, 2020-12-06:
Hiya,

Not moved Kinect from the USB port where it has worked for months.

Not really sure what you mean by
ust to be sure this problem isn't related to obs-plugin, this issue happens also with the official SDK examples

Jenko1, 2020-12-06:
Hi all, just tried the Kinect on another machine with OBS installed and it works correctly, so guessing it’s a usb issue, any ideas how to rectify this.

Jenko1, 2020-12-06:
Think I’ve managed to sort it now, went to setting and connected devices. Removed Kinect from within there, unplugged and plugged back in and all now seems fine and working like before. Thanks for all your help. Stay safe.

SirLynix, 2020-12-07:
It happens, I had some USB troubles myself and yeah, basically trying to plug thing in other ports is a fix. Glad it was not an hardware issue after all.

closed 2020-12-07 by SirLynix