SirLynix/obs-kinect #41: Kinect doesn't activate in portable mode

VeGas-Gh0st, 2020-12-08:
in regular obs the plugin works fine, but in portable mode i cant get the camera to activate even in admin mode.

SirLynix, 2020-12-08:
This is weird since I run all my tests in portable mode without any issue.

What does the logs say?
Have you tried with the last release (0.3RC2)?

VeGas-Gh0st, 2020-12-08:
ok did more testing, v3 overwrote something im thinking as this one now works fine.

[image]

Issue resolved it seems, thank you for your quick response.

SirLynix, 2020-12-09:
After looking a bit closer it seems you were having 0.3 in normal mode and 0.2 in portable mode. As you're using a KinectV1 only 0.3 does support it.

I know 0.3 is kinda WIP for now but what issues are you talking about?

SirLynix, 2020-12-11:
Closed for no answer.

closed 2020-12-11 by SirLynix