SirLynix/obs-kinect #43: FPS lagging

Fyrhil, 2020-12-19:
Hello ! 
I followed all the instructions , my Kinect v2 is well detected but unfortunately the camera is around 10-15fps. 
I tried with 0.2 0.3rc1 and 03rc2 and it’s the same (more fluid with the 0.3). 
My setup is Ryzen 7 3700X , 2070S and 16Gb Ram at 3100Mhz. I don’t think the problem is from my hardware but I messed up somewhere. 
Thanks for you help ^^

SirLynix, 2020-12-19:
Hi!

Is the Kinectv2 connected to a USB3 port? The problem may be coming from the USB bandwidth.

More importantly, is your kinect working as expected with the Kinect v2 samples? (try the color basic one in C++).

You can also try to use latest build from #7 (end of topic) to adjust the color settings, the Kinectv2 stays at 15fps when it's too dark by default.

Fyrhil, 2020-12-19:
Yes it's connected to an usb 3 port ! About the Kinect sample in C++ do I have something to download in Kinect SDK ? (Abuse I can't download anything on it idk why)

SirLynix, 2020-12-19:
Once you have the Kinect SDK it should be installed already. Try starting the "SDK Browser V2.0 (Kinect for Windows)" app and run the Color Basic sample.
[image]

Fyrhil, 2020-12-19:
First I want to thank you for your fast answer it's really appreciated ^^ 
I'll join some photos about the problem I have with Kinect SDK but I'll understand if it's not your thing because some things aren't working on it and I can't run any program [image][image]

SirLynix, 2020-12-19:
Since you are french, autant continuer en français. 😄 

Pour le coup on dirait que le SDK est mal installé. Essaie la dernière version de #7 pour voir si tu peux changer ça avec les paramètres de couleur. Vérifie aussi la charge OBS pour voir ce qui ralentit.

Fyrhil, 2020-12-19:
Oh quel bonheur ahah ! Alors voilà j'ai essayé de passer sur le version 0.3 comme le dit le #7 enfin si j'ai bien compris , je te joins la charge de OBS vu que je peux pas partager de vidéo pour te montrer ce que ça donne :( [image]

SirLynix, 2020-12-19:
Tu as bien pris la dernière version ? ([**obs-kinect-camera-controls.zip**](https://github.com/SirLynix/obs-kinect/files/5698771/obs-kinect-camera-controls.zip)).

Ouvre les statistiques d'OBS, tu devrais avoir quelque chose comme ça:
[image]

Fyrhil, 2020-12-19:
Écoute je viens de le mettre il y a du mieux mais on est toujours pas fluide fluide ahah ! Je t'ai fais un screen des stars je sais pas dis ça va t'aider plus que ça [image]

SirLynix, 2020-12-19:
Essaie de jouer avec les paramètres d'exposition, c'est bien possible que ça soit causé par un environnement sombre qui nécessite une exposition plus longue.
Est-ce que la KinectV2 est détectée comme webcam sur ton système ? Est-ce que tu peux l'utiliser dans une application ? (Caméra / Discord / Facebook / etc.).
L'idée ici est surtout de voir si le problème vient du plugin ou d'autre chose. En tout cas je ne vois rien de problématique dans les paramètres OBS.
Tiens j'y pense, tu as jeté un oeil aux logs d'OBS ? (Aide => Fichier Journaux => Voir le journal actuel) Est-ce qu'il y a des lignes qui concernent obs-kinect ?
Est-ce que tu as essayé un autre port USB pour voir si le problème se produit aussi ? (La KinectV2 est très gourmande en bande-passante USB).

Fyrhil, 2020-12-20:
Bah écoute ce matin j'ai tenté de réinstaller le SDK, je peux maintenant run les programmes , et j'ai changé de port USB et ça marche nickel , j'ai un beau 30fps sur la cam je suis refais ! Donc je sais pas d'où ça venait mais ça a l'air de marcher maintenant ! Par contre avant je pouvais l'utiliser en tant que webcam sur d'autres logiciels , maintenant je ne peux plus. Mais au moins ça marche sur obs donc bon. 
Question bonus , j'ai eu l'impression que des gens faisait tourner la kinect a 60fps en tant que webcam je sais pas si j'ai rêvé.

SirLynix, 2020-12-20:
Bon ben tant mieux 😄 

Pour ce qui est du 60 FPS de ce que j'en sais ce n'est pas possible, dû à la combinaison du capteur couleur et du capteur infrarouge. Mais si tu vois des projets qui sont capables d'exploiter les 60 FPS sur une Kinect n'hésite pas à m'en parler !

Question bonus pour moi aussi, tu as pu tester les paramètres de couleur sur la Kinect v2 ? J'ai très peu de retour sur cette dernière build malheureusement 😞

Fyrhil, 2020-12-20:
Tu veux dire tout cette partie la ? [image]

SirLynix, 2020-12-20:
Ouaip. Ça manque encore un peu de traduction mais je parle bien de celle-là.

Fyrhil, 2020-12-20:
Franchement je trouve ça nickel , ça fonctionne bien , les réglages sont faciles à prendre en main , tu peux faire des effets sympas avec la balance des blancs en auto , j'ai pas compris par contre le réglage pour changer la luminosité des leds 😆 mais pourquoi pas ! 
J'ai eu qu'un seul soucis c'est avec l'effet fond vert mais en mode flou d'arrière plan , j'ai eu 2 crash et après ça a fonctionné de nouveau normalement ^^

SirLynix, 2020-12-20:
Si tu peux m'envoyer les crashlogs ça m'intéresse, pour essayer de corriger ça pour la prochaine release !

Fyrhil, 2020-12-20:
[Crash 2020-12-20 11-40-05.txt](https://github.com/SirLynix/obs-kinect/files/5720876/Crash.2020-12-20.11-40-05.txt)
[Crash 2020-12-20 11-40-26.txt](https://github.com/SirLynix/obs-kinect/files/5720877/Crash.2020-12-20.11-40-26.txt)

Les voilas ^^

SirLynix, 2020-12-20:
Bon ça a l'air de crasher dans ton driver, je ne pense pas pouvoir y faire grand chose en l'état.
Tant que tu arrives à faire marcher le plugin c'est good pour moi.

closed 2020-12-20 by SirLynix