SirLynix/obs-kinect #45: Motion Tracking Support

sarvepalli18, 2021-01-08:
When i add kinect as a source in OpenNIVirtualCam i can see the motion tracking when smart tracking feature is selected. I do not see the same with OBS though, this is very helpful feature especially when there are multiple people in the room or when the subject is moving. Would be a great addition to the already available excellent features in obs kinect :)

SirLynix, 2021-01-08:
Hi.

What exactly is that motion tracking feature? I installed OpenNIVirtualCam to try it out and didn't see any change when enabling Smart Tracking.

SirLynix, 2021-01-08:
Okay I managed to make it work, and took a look at the code, it's actually a face tracking, centering on someone face, it should be pretty easy to do 😄

sarvepalli18, 2021-01-11:
Yes it is face tracking. I have implemented face tracking using particles filters but not sure how to implement it using kinect APIs. I am a noob though, but very much interested in Computer Vision.

Also one more thought i had is that since kinect receives audio, it will be a nice feature to track audio, like whoever is the speaker in the room the camera focuses on the speaker.

SirLynix, 2021-01-12:
Yes I've seen that KinectV2 (and maybe v1) have directional audio. Having that and face tracking would be a really nice feature indeed.
I can't promise I'll do it before releasing 1.0 (I should have released it a while ago already), I still have to work on the Linux version to release it.