SirLynix/obs-kinect #46: I finished First build of korean language

yeonggille, 2021-01-09:
The .ini file is inside of the zip. Happy new year! and I hope there will be happiness who seeing this follows.
[KOR_LANG_MK_1.zip](https://github.com/SirLynix/obs-kinect/files/5790418/KOR_LANG_MK_1.zip)

SirLynix, 2021-01-09:
Thanks you! I will include this in the 0.1 release 😄

closed 2021-01-09 by SirLynix

yeonggille, 2021-01-15:
Tell me if there is encoding or other error when you test it out or during commiting the file!

SirLynix, 2021-01-15:
I commited it already, seems good 😄 (https://github.com/SirLynix/obs-kinect/commit/0ec5bae459ac47bc7da13ad6a9f096222c41bd80)

yeonggille, 2023-06-06:
Hello, Long Time No See, I hope you doing well. I want to check my korean strings are working fine / Or Main English String have changes since 2 years ago, as I sold Kinect recently and maintain the legacy Korean String code

SirLynix, 2023-06-06:
Hi!

Yes, you can check the current translation here: https://github.com/SirLynix/obs-kinect/blob/master/data/obs-plugins/obs-kinect/locale/ko-KR.ini

You will see there are a few missing translation Korean (the ones in english), not much though.

commit 0ec5bae4 (2021-01-09): Add ko-KR locale (close #46)