SirLynix/obs-kinect #47: Azure Kinect Files in the OBS folder.

caseyfarina, 2021-01-12:
I've been using this last year with a Kinect V2 an it worked great? Thank you!

I recently switched to an Azure Kinect and the kinect source shows up in OBS but is totally blank. Does the entire Azure Kinect SDK need to be copied to the OBS directory?
 I put the folder labeled "Azure Kinect SDK v1.4.1" in the 

obs - bin - 64bit directory and that does not seem to be working. 

Thanks you!

SirLynix, 2021-01-12:
Hi.

What version did you install? The most recent released one (0.3 RC2) doesn't support Azure Kinect.
The most recent release of obs-kinect (which adds a lot of thing) is currently unreleased but can be downloaded at the end of #7, all dll are included.

caseyfarina, 2021-01-12:
Oh, I see. It's working great now! This is a great asset for those of us who don't have enough room for a full green screen. The noise reduction looks much better on the Azure Kinect vs the Kinect V2. Do you have a patreon? How can I financially support this?

SirLynix, 2021-01-12:
Glad to hear it's working well for you! And yes the depth quality of the Azure Kinect is so much better (I'd like to add some processing on that to improve the result, see #32).

Thank you for wanting to support me, I'm thinking of opening a patreon or something like that but I'm waiting to release some other projects before. However if you wish to financially support this project you can click on the "Sponsor" button at the top and either use GitHub sponsor (a sort of patreon like for developpers) or Paypal directly 😄

SirLynix, 2021-01-12:
I'm closing this issue as it's resolved. Thank you for your support 😄

closed 2021-01-12 by SirLynix