SirLynix/obs-kinect #50: No Linux support despite Microsoft support on the Azure Kinect

Utopiah, 2021-01-16:
Creating the issue so that whenever testing is needed I can provide some feedback. Thanks for the plugin already.

SirLynix, 2021-02-11:
The first Linux binaries including Azure Kinect support can be found at the end of this thread: #9.

Utopiah, 2021-02-11:
Thanks, I did make sure my Kinect was working well then rebuild obs from source but I'll keep the actual install and test for tomorrow. I admit I'm very confused in the directory structure and location of obs plugins in general and the 2 zip files in the other issue but it might be just the end of the day. Thanks for keeping me posted and hopefully I'll manage to give it a go tomorrow.

SirLynix, 2021-02-11:
Yes it may be quite confusing for now, making it easier to install is one of my next task. 😄 
Could you run `ldd` on the obs-kinect-azuresdk.so binary? Just in case you may be missing some dependencies.

Utopiah, 2021-02-16:
```
fabien@fabien-CORSAIR-ONE-i160:~/Prototypes/obs-kinect$ ldd libobs-kinect-azuresdk.so
        linux-vdso.so.1 (0x00007ffd4a10a000)
        libobs-kinect.so => not found
        libobs.so.0 => /lib/libobs.so.0 (0x00007f87bab5c000)
        libk4a.so.1.4 => not found
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f87ba97b000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f87ba960000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f87ba76e000)
        libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007f87ba71b000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f87ba715000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f87ba5c6000)
        libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007f87ba489000)
        libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f87ba45f000)
        libX11-xcb.so.1 => /lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007f87ba45a000)
        libpulse.so.0 => /lib/x86_64-linux-gnu/libpulse.so.0 (0x00007f87ba403000)
        libavcodec.so.58 => /lib/x86_64-linux-gnu/libavcodec.so.58 (0x00007f87b8e46000)
        libjansson.so.4 => /lib/x86_64-linux-gnu/libjansson.so.4 (0x00007f87b8e37000)
        libavformat.so.58 => /lib/x86_64-linux-gnu/libavformat.so.58 (0x00007f87b8bbf000)
        libavutil.so.56 => /lib/x86_64-linux-gnu/libavutil.so.56 (0x00007f87b8a96000)
        libswscale.so.5 => /lib/x86_64-linux-gnu/libswscale.so.5 (0x00007f87b8a01000)
        libswresample.so.3 => /lib/x86_64-linux-gnu/libswresample.so.3 (0x00007f87b89dd000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f87b89c1000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f87b899e000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f87bae82000)
        libsystemd.so.0 => /lib/x86_64-linux-gnu/libsystemd.so.0 (0x00007f87b88ef000)
        libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007f87b88e9000)
        libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f87b88df000)
        libpulsecommon-13.99.so => /usr/lib/x86_64-linux-gnu/pulseaudio/libpulsecommon-13.99.so (0x00007f87b885d000)
        libvpx.so.6 => /lib/x86_64-linux-gnu/libvpx.so.6 (0x00007f87b8625000)
        libwebpmux.so.3 => /lib/x86_64-linux-gnu/libwebpmux.so.3 (0x00007f87b841b000)
        libwebp.so.6 => /lib/x86_64-linux-gnu/libwebp.so.6 (0x00007f87b81b2000)
        liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f87b8187000)
        librsvg-2.so.2 => /lib/x86_64-linux-gnu/librsvg-2.so.2 (0x00007f87b785c000)
        libgobject-2.0.so.0 => /lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f87b77fc000)
        libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f87b76d3000)
        libcairo.so.2 => /lib/x86_64-linux-gnu/libcairo.so.2 (0x00007f87b75b0000)
        libzvbi.so.0 => /lib/x86_64-linux-gnu/libzvbi.so.0 (0x00007f87b7521000)
        libsnappy.so.1 => /lib/x86_64-linux-gnu/libsnappy.so.1 (0x00007f87b7514000)
        libaom.so.0 => /lib/x86_64-linux-gnu/libaom.so.0 (0x00007f87b7076000)
        libcodec2.so.0.9 => /lib/x86_64-linux-gnu/libcodec2.so.0.9 (0x00007f87b6292000)
        libgsm.so.1 => /lib/x86_64-linux-gnu/libgsm.so.1 (0x00007f87b6283000)
        libmp3lame.so.0 => /lib/x86_64-linux-gnu/libmp3lame.so.0 (0x00007f87b620b000)
        libopenjp2.so.7 => /lib/x86_64-linux-gnu/libopenjp2.so.7 (0x00007f87b61b5000)
        libopus.so.0 => /lib/x86_64-linux-gnu/libopus.so.0 (0x00007f87b6154000)
        libshine.so.3 => /lib/x86_64-linux-gnu/libshine.so.3 (0x00007f87b5f49000)
        libspeex.so.1 => /lib/x86_64-linux-gnu/libspeex.so.1 (0x00007f87b5f2b000)
        libtheoraenc.so.1 => /lib/x86_64-linux-gnu/libtheoraenc.so.1 (0x00007f87b5eee000)
        libtheoradec.so.1 => /lib/x86_64-linux-gnu/libtheoradec.so.1 (0x00007f87b5ece000)
        libtwolame.so.0 => /lib/x86_64-linux-gnu/libtwolame.so.0 (0x00007f87b5ea8000)
        libvorbis.so.0 => /lib/x86_64-linux-gnu/libvorbis.so.0 (0x00007f87b5e78000)
        libvorbisenc.so.2 => /lib/x86_64-linux-gnu/libvorbisenc.so.2 (0x00007f87b5dcd000)
        libwavpack.so.1 => /lib/x86_64-linux-gnu/libwavpack.so.1 (0x00007f87b5da1000)
        libx264.so.155 => /lib/x86_64-linux-gnu/libx264.so.155 (0x00007f87b5ae3000)
        libx265.so.179 => /lib/x86_64-linux-gnu/libx265.so.179 (0x00007f87b4b75000)
        libxvidcore.so.4 => /lib/x86_64-linux-gnu/libxvidcore.so.4 (0x00007f87b4a62000)
        libva.so.2 => /lib/x86_64-linux-gnu/libva.so.2 (0x00007f87b4a37000)
        libxml2.so.2 => /lib/x86_64-linux-gnu/libxml2.so.2 (0x00007f87b487d000)
        libbz2.so.1.0 => /lib/x86_64-linux-gnu/libbz2.so.1.0 (0x00007f87b486a000)
        libgme.so.0 => /lib/x86_64-linux-gnu/libgme.so.0 (0x00007f87b481d000)
        libopenmpt.so.0 => /lib/x86_64-linux-gnu/libopenmpt.so.0 (0x00007f87b4630000)
        libchromaprint.so.1 => /lib/x86_64-linux-gnu/libchromaprint.so.1 (0x00007f87b4618000)
        libbluray.so.2 => /lib/x86_64-linux-gnu/libbluray.so.2 (0x00007f87b45c7000)
        libgnutls.so.30 => /lib/x86_64-linux-gnu/libgnutls.so.30 (0x00007f87b43f1000)
        libssh-gcrypt.so.4 => /lib/x86_64-linux-gnu/libssh-gcrypt.so.4 (0x00007f87b4365000)
        libva-drm.so.2 => /lib/x86_64-linux-gnu/libva-drm.so.2 (0x00007f87b4360000)
        libva-x11.so.2 => /lib/x86_64-linux-gnu/libva-x11.so.2 (0x00007f87b4358000)
        libvdpau.so.1 => /lib/x86_64-linux-gnu/libvdpau.so.1 (0x00007f87b4350000)
[... cut ...]
        libkrb5support.so.0 => /lib/x86_64-linux-gnu/libkrb5support.so.0 (0x00007f87b1027000)
        libblkid.so.1 => /lib/x86_64-linux-gnu/libblkid.so.1 (0x00007f87b0fd0000)
        libpcre2-8.so.0 => /lib/x86_64-linux-gnu/libpcre2-8.so.0 (0x00007f87b0f40000)
        libdatrie.so.1 => /lib/x86_64-linux-gnu/libdatrie.so.1 (0x00007f87b0f36000)
        libgraphite2.so.3 => /lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f87b0f09000)
        libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007f87b0f00000)

```

SirLynix, 2021-02-16:
Thank you, it seems the only dependency you will be missing is Azure Kinect SDK, which you can install by following the instructions [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md)

Utopiah, 2021-02-16:
Actually I already built from sources in `~/Prototypes/Azure-Kinect-Sensor-SDK/build/bin/` after following https://tianyusong.com/2019/11/13/how-to-install-azure-kinect-sdk-on-ubuntu-16-04/ because the Microsoft package repositories are... well a damn mess. If possible I'd prefer relying on those because I tried again from the official installation instructions but without success.

SirLynix, 2021-02-24:
That will not be a problem. However you'll still need to get the proprietary depth engine from Microsoft if you wish to use the depth/infrared streams or effects.

closed 2023-01-20 by SirLynix