SirLynix/obs-kinect #54: Hey! Is there a way to to make it so that with the greenscreen effect there is no minimum distance required?

Lenr0k, 2021-04-08:
Currently the greenscreen effect makes everything thats under the kinects detection range transparent. Is there a way or planned feature that would make it so that stuff outside its detection range doesn't get keyed?

SirLynix, 2021-04-08:
Hello.

Each Kinect model has a minimum depth range below which it cannot read depth (tell how far the pixel is), theses values are reported as invalid (exactly the same as objets too far away from the sensor, or in "depth shadow"), this is an hardware limitation and I cannot do anything about it.

The only thing I can do is to try to filter out the result, which isn't an easy task to do (especially in realtime).

Lenr0k, 2021-04-08:
hm.

Would an option to make invalid value depths not be keyed be possible? Or would it just cause problems?

SirLynix, 2021-04-08:
Considering invalid depth values in the greenscreen effect would look *very* bad, as it would include the flickering background/foreground as well. Which is why I didn't leave this option in the plugin.

Lenr0k, 2021-04-08:
Yeah thats understandable. Guess i'll have to put the kinect a little bit further away then.

Thanks!

closed 2021-04-08 by SirLynix

SirLynix, 2025-08-24:
> also couldnt the faux-greenscreen use the ir camera for depth instead of the weird calculated depth thing.

I guess it could, but how? Except filter based on luminance (which would probably look bad as it's affected by reflectivity).

Depth is derived from IR already.

home-gihub, 2025-08-25:
> > also couldnt the faux-greenscreen use the ir camera for depth instead of the weird calculated depth thing.
> 
> I guess it could, but how? Except filter based on luminance (which would probably look bad as it's affected by reflectivity).
> 
> Depth is derived from IR already.

Yea, I know it would look bad but something looking bad is better than nothing...

SirLynix, 2025-08-25:
Actually you can already do this using OBS native filters.

[image]

Here's a try using a Kinect Source configured to show the infrared feed, and a Luma key filter filtering pixels based on luminance.

Hope it gives you good results.

home-gihub, 2025-08-25:
oh i did this like two hours ago nvm