SirLynix/obs-kinect #55: OBS - Kinect Source Stuttering

DeNo64, 2021-04-29:
Hey mate, 

I'm having an issue in OBS where every ~3 seconds the Camera will stutter (freeze for ~0.5 seconds).

Color Basics on the SDK works fine. This seems to only happen on OBS.

I have Runtime v2.2, SDK v2.0, plugin 0.3 RC2. There also doesn't seem to be any excessive resource utilisation from the plugin/OBS.

You're assistance would be much appreciated.

SirLynix, 2021-04-29:
Hi.

Are you using only the color output in your Kinect Source?  
Is there any relevant log in OBS Studio related to those freeze (or obs-kinect in general)?  
Is your Kinect plugged in a USB 3 port?  
Have you tried other SDK examples (which do imply color and depth)?

And last but not least, does it still happen with the [nightly version](https://github.com/SirLynix/obs-kinect/issues/7#issuecomment-800903760)?

DeNo64, 2021-04-29:
1. Exactly the same on each of the three outputs.
2. Noticed these logs (below)
3. Plugged into a USB 3.1 on my Motherboard
4. I've tested other SDK Examples and the motion is all smooth

5. I'll give the nightly version a go and let you know

15:35:16.103: LoadLibrary failed for '../../obs-plugins/64bit/KinectBackgroundRemoval180_64.dll': The specified module could not be found.
15:35:16.103:  (126)
15:35:16.103: Module '../../obs-plugins/64bit/KinectBackgroundRemoval180_64.dll' not loaded
...
15:35:16.118: LoadLibrary failed for '../../obs-plugins/64bit/obs-kinect-sdk10.dll': The specified module could not be found.
15:35:16.118:  (126)
15:35:16.118: Module '../../obs-plugins/64bit/obs-kinect-sdk10.dll' not loaded

SirLynix, 2021-04-29:
Those errors are just because the Kinect SDK v1.8 is not installed, you don't need it if you don't want to support Kinect v1 (for xbox 360).

How is the CPU and GPU usage while running it? Does running OBS-Studio with admin privileges help?

DeNo64, 2021-04-29:
Hmmm this is interesting

OBS was trying to access images, for my scenes, on a secure network drive. Once I gave my computer access to the network drive, and OBS could access the images, the stuttering issue went away.

So for whatever reason OBS continuously attempting to access these inaccessible files cause the stutter.

Thanks for the quick assist, mate. Keep up the great work on the plugin.

SirLynix, 2021-04-29:
Glad to hear it's fixed. Hope the plugin serves you well!

closed 2021-04-29 by SirLynix