SirLynix/obs-kinect #56: Kinect V1 device detected but remains black screen

RogueMissile, 2021-06-11:
I have the plugin installed into OBS V27, using 0.3 release candidate 2 and I added the plugin to my sources and moved it to the top. All of the properties are there, and my device is detected as KinectSDK1.0 - Kinect #0: USB along with the device serial. 

No matter what settings I change though nothing happens and the Kinect doesn't display any image. Yet it works perfectly fine within the dev toolkit apps.

RogueMissile, 2021-06-11:
Edit: I had an app still running from the toolkit taking control and it works after closing that app. Sorry.

closed 2021-06-11 by RogueMissile

SirLynix, 2021-06-11:
Yes, for the KinectV1 at most one app can use it at a given time. I'm glad you found the solution.
Be aware that Microsoft examples may bug some time, like stalling after closing the window and not releasing the Kinect properly (if that happens all you have to do is to kill the .exe via the task manager).