SirLynix/obs-kinect #57: Kinect 1 - disables high res mode when changing scenes

abelbour, 2021-07-16:
Doesn't save the high res option. When you return to a scene with KinectSource or open OBS high res mode is disabled, even though the option is checked. Unchecking and checking again enables it until you change the scene again.

SirLynix, 2021-07-17:
I'll take a look at that

SirLynix, 2021-07-17:
@abelbour I can't reproduce it.

I have a scene with my KinectSource (with high res), and another scene without anything, switching between the two doesn't reproduce the issue (my KinectSource is still high res when I get back on its scene). Closing and restarting OBS doesn't seem to reproduce the issue either.

To be clear, I'm checking my KinectSource is high res by looking at its resolution (1280x960).

So maybe it's a bug a little bit more complicated (do you have another KinectSource on your other scenes) or maybe it's already fixed in the last version, which version of obs-kinect are you using?

abelbour, 2021-07-17:
@SirLynix  I'm using OBS 27.0.1 64-bit and obs-kinect 0.3-rc2. I have a Kinect 360.
Also, when I leave the scene, the elevation resets to 0°, and when I come back to the scene, go to settings and turn high-res off and on again the elevation returns to the one saved before.
Thanks for your time!

abelbour, 2021-07-17:
Also, If I leave the properties dialog for KinectSource open I can change scenes without it reseting to lo-res or changing the elevation. Thanks again.

SirLynix, 2021-07-17:
I fixed a bug related to elevation when investigating for your issue.
From what you describe it looks like obs parameters are not properly applied when switching from one scene to the other, which is weird since this is what I tested.

Could you try using the [latest build artifact](https://github.com/SirLynix/obs-kinect/suites/3263076291/artifacts/75940695) to see if it still occurs with the last version?

If yes, could you try making a new scene collection and reproduce the bug using an empty scene and a scene containing a KinectSource? (this is what I used to try to reproduce it).

abelbour, 2021-07-17:
@SirLynix now it works! I tried the new build with a new scene collection as you asked and worked ok, the settings persisted even after closing and reloading OBS. Then I tried it with the scene collection I used before and it worked ok too, both resolution and elevation settings persist. Thanks again!!

SirLynix, 2021-07-17:
I'm glad it works for you, so this is probably a bug that was fixed between 0.3-rc2 and the master branch.
I will soon get back to work on obs-kinect (to fix linux support and add a few features) and release a new version.

closed 2021-07-17 by SirLynix

abelbour, 2021-07-17:
I'm glad to help you testing If you need it. BTW SDK background removal is not working for me, I guess it must be related to my setup. I'm not using it anyway but I can test it if you need. Thanks!

SirLynix, 2021-07-17:
Not working for you like, is not available? Or you mean it just doesn't show anything?

If it doesn't show anything try to stand a bit farther from the Kinect, it has to recognize your body for it to work.

SirLynix, 2021-07-18:
@abelbour I tried to use SDK background removal and discovered it didn't work with high-res mode, because it wasn't made to handle this size of image. I don't think I can do much about it (well except resize the image, handle it to Microsoft SDK for background removal and then apply alpha values on the high-res image, but I'm not sure this would be worth it).

I never tried to use both at the same time so you've discovered a limitation of obs-kinect.