SirLynix/obs-kinect #58: Control Brightness, Hue, or Gain

Wolfhound905, 2021-10-20:
If possible could you add the ability to adjust the brightness/iso of the Kinect via the SDK? I try to use the filter in obs but since the Kinect itself is too bright the data is gone so I can't just lower brightness.

[image]

SirLynix, 2021-10-21:
Hi,

What version of obs-kinect are you using and what Kinect do you have?

Nightlies versions of obs-kinect have color control available for all Kinect models that support it

closed 2022-04-18 by SirLynix

Wolfhound905, 2022-05-16:
Do you know a way I can adjust the exposure on the sensor level? Because adjusting it with OBS controls is useless because the overexposed areas are just white, there is no data there.

SirLynix, 2022-05-16:
With the last obs-kinect version you should be able to change the exposure of all Kinect sensors, directly from the obs-kinect source.