SirLynix/obs-kinect #59: Kinect not showing in OBS 27.1.3 64bit windows

Supermankal718, 2022-02-08:
I have downloaded the obs plugin, the github plugin, windows runtime. and windows SDK like it said in [this video](https://www.youtube.com/watch?v=iEtKBZ2crUE&t) but I still have no luck with the kinect showing up as a source in OBS.

Im able to get it to come up in runtime and the sdk.

SirLynix, 2022-02-08:
Hi.

Have you tried to download and install the last release (0.3)?
Have you checked the [commonly asked questions](https://github.com/SirLynix/obs-kinect#commonly-asked-questions) part of the guide?

What Kinect model are you using?

Supermankal718, 2022-02-09:
Yes I just copied the files over from the 0.3 zip into the obs files. Restarted my pc and still doesnt show up as a source. 

Im using the xbox one kinect.

SirLynix, 2022-02-09:
Could you check your OBS logs to see if there any notice mentioning
obs-kinect?

Supermankal718, 2022-02-09:
went to show log files and went into the latest one and searching for "obs-kinect" shows cannot find obs-kinect

SirLynix, 2022-02-09:
Do you have a obs-kinect.dll in your obs-studio/obs-plugins/64bits folder? Are you running 32bits version?

Even if some dependencies are missing, OBS should list obs-kinect.dll in the logs at loading time

Supermankal718, 2022-02-09:
just checked and running 64 bit obs and the file is in that folder.
[image]

SirLynix, 2022-02-09:
Could you send me your OBS log?

Supermankal718, 2022-02-12:
https://obsproject.com/logs/1_9FhGgt7M9e32Cb

StellarVision2099, 2022-02-15:
i seem to be having a similar problem. updated to OBS 27.2.0 today and now obs-kinect plugin no longer finds the kinect. can still see it through regular video capture source but obv no green screen functionality that way. ill drop my log here for more info.
[2022-02-14 18-14-10.txt](https://github.com/SirLynix/obs-kinect/files/8064812/2022-02-14.18-14-10.txt)

SirLynix, 2022-02-15:
This is weird, but at least your OBS loads obs-kinect, which isn't the case for @Supermankal718 (it looks like obs-kinect is simply not there).

Could both of you try the [nightly version](https://github.com/SirLynix/obs-kinect/suites/4952769399/artifacts/143797011)?

williethewolf, 2022-02-15:
Same issue here. I just updated to 27.2.0 and the Kinect source shows the Kinect Device drop down as "No device".

williethewolf, 2022-02-15:
I tried the Nightly Version and seems to be working. On startup, the Device drop down is still empty but you can now select Kinect V2 as the device and seems to work as intended.

williethewolf, 2022-02-15:
I did get a crash while playing with the blur slider with the camera on Blur Background mode. Not sure if it was strictly OBS related but here's the log
[Crash 2022-02-15 12-41-12.txt](https://github.com/SirLynix/obs-kinect/files/8074551/Crash.2022-02-15.12-41-12.txt)
.

StellarVision2099, 2022-02-15:
thank you! it is working now.

SirLynix, 2022-02-16:
Glad to hear the nightly version is working. I'm quite busy at the moment but I'd like to release it as obs-kinect 1.0 very soon. Don't hesitate to post here if something doesn't work right or can be improved. About your crash @williethewolf it seems to be coming from the GPU driver, so I'm not sure if it's obs-kinect fault.

Supermankal718, 2022-02-16:
> This is weird, but at least your OBS loads obs-kinect, which isn't the case for @Supermankal718 (it looks like obs-kinect is simply not there).
> 
> Could both of you try the [nightly version](https://github.com/SirLynix/obs-kinect/suites/4952769399/artifacts/143797011)?

Still isnt working for me. Im going to do a fresh install of obs and see if itll work.

Dennarb, 2022-02-25:
> This is weird, but at least your OBS loads obs-kinect, which isn't the case for @Supermankal718 (it looks like obs-kinect is simply not there).
> 
> Could both of you try the [nightly version](https://github.com/SirLynix/obs-kinect/suites/4952769399/artifacts/143797011)?

This link does not appear to be working for me, but I also seem to be having the same issue as others after upgrading to OBS 27.2.1

SirSyao, 2022-02-27:
Same here obs 27.2.1, with the nightly i only have chrashes, but with the 0.2 release work just fine

SirLynix, 2022-02-28:
@SirSyao Could you post your crashlogs with the nightly version?

SirSyao, 2022-03-01:
here you go
[Crash 2022-02-28 18-19-14.txt](https://github.com/SirLynix/obs-kinect/files/8157824/Crash.2022-02-28.18-19-14.txt)

SirLynix, 2022-03-01:
@SirSyao Could you try to remove the obs-kinect-freenect DLL (1 and 2) from your plugin folder?

abbaselmas, 2022-03-01:
OBS 27.2.1 here, nightly build resolved my issue too. Kinect v1

SirLynix, 2022-03-01:
Hi @MattFoxall, did you try to use the [nightly version](https://github.com/SirLynix/obs-kinect/suites/5471850177/artifacts/174363259)?

MattFoxall, 2022-03-01:
> Hi @MattFoxall, did you try to use the [nightly version](https://github.com/SirLynix/obs-kinect/suites/5471850177/artifacts/174363259)?

Yeah i did it just does the same thing as posted above

SirLynix, 2022-03-01:
Ok, then it must be some kind of issue between your Kinect and your computer. Have you checked OBS logs? Have you tried the official Microsoft examples from their Kinect SDK (links are in the [README](https://github.com/SirLynix/obs-kinect#how-to-build-for-people-wanting-to-contribute)).

MattFoxall, 2022-03-01:
@SirLynix  thank you for the suggested help, But im not a very technical person is there any possible way we could talk on like discord and help me through that instead ?

SirLynix, 2022-03-01:
Sure, feel free to add me: Lynix#3516

SirSyao, 2022-03-01:
> @SirSyao Could you try to remove the obs-kinect-freenect DLL (1 and 2) from your plugin folder?
@SirLynix there are none freenect.dll on the plugins folder, only in the bin folder, i delete all and for now nothing crash

[image]

Pyrdy, 2022-03-12:
Hi, I have same problem. Also cannot get the Nightly version to test, perhaps it's removed? Problems happened when upgraded to 27.2.3 version of OBS Studio.Downgrading to 27.1.1 fixes the Kinect, but cannot connect to Twitch anymore. Quote that I found: "As mentioned above users are getting this error when using an older version of the OBS software. Twitch API v5 is no longer available for use from 28th February 2022. Hence you need to upgrade your OBS version in order to not get the above issue." Hope this helps. Lookin forward to the 1.0 version, and I will put some support to you as well. BR, Riku

SirLynix, 2022-03-12:
I released the current version so you can download it here: https://github.com/SirLynix/obs-kinect/releases/tag/v1.0-rc1

Pyrdy, 2022-03-12:
> I released the current version so you can download it here: https://github.com/SirLynix/obs-kinect/releases/tag/v1.0-rc1

Thanks man. Seems to work now with new OBS version. Now I can continue use it to stream me dancing my fat belly away. :-)

Here is sample of my overlay  there. 
[image]

BR, Riku

closed 2022-03-12 by SirLynix