SirLynix/obs-kinect #62: xmake fails to build on linux

weirdal3333, 2022-03-31:
[image]

What could be the cause? This is a fresh linux installation, and I followed the build directions exactly.

The latest release has no information on where to put the files. Where do they go once they're built?

SirLynix, 2022-03-31:
Hi.

XMake will handle every obs-kinect dependency, but you still need to clone and build obs-studio yourself (check [the CI](https://github.com/SirLynix/obs-kinect/blob/master/.github/workflows/linux-build.yml#L110) for help)

You need to write a config.lua file next to the xmake.lua file, to tell it where your obs-studio build folder is located.
See the [template](https://github.com/SirLynix/obs-kinect/blob/master/config.lua.default) and what the [CI](https://github.com/SirLynix/obs-kinect/blob/master/.github/workflows/linux-build.yml#L146) does].

Files are built in the bin/linux_x86_64_releasedbg folder by default, I'm not really sure where they should be on Linux.

weirdal3333, 2022-03-31:
I followed the CI to the best of my ability, but still ended up with a package folder at the end.
I guess the main issue here is that we don't know where to put the files generated.
Where do they go? It should be the same location across most linux distros.

weirdal3333, 2022-03-31:
I am pretty confused with his setup.
I successfully compiled the obs-kinect repo, and I get the same files in the latest release. These files are in a completely different folder structure from what @jugarconfuego has in his setup.

SirLynix, 2022-03-31:
The folder structure of obs-kinect packaging follows what is needed on Windows, since this is the platform I'm using. obs-kinect files should probably be moved to other folder.

weirdal3333, 2022-03-31:
Going back to the main issue, The readme makes it sound like you can just run xmake and everything will just work, including the building of all dependencies (that would include obslib, logically)

I'll close this issue since it actually does work after deendencies are in place, but an updated readme is definitely in order.
Maybe setting up the wiki function of github would be a good idea for installation instructions for different operating systems.

closed 2022-03-31 by weirdal3333

SirLynix, 2022-03-31:
You're right, this isn't very clear. I should also make a xmake package that builds libobs. I will update the README.