SirLynix/obs-kinect #63: Kinect not found in device list

JawShoeAaa, 2022-04-09:
No matter how many times i refresh the list my working, plugged in and connected Kinect isn't displayed on OBS for me to select , it can however function in the SDK Browser on any of it's programs, and I've copied both folders into my OBS and been sure to restart

Can anyone assist in this problem with OBS

SirLynix, 2022-04-09:
Hi.
Could you show me your OBS logs?

JawShoeAaa, 2022-04-09:
[image]
like this?

SirLynix, 2022-04-09:
No, I mean OBS text logs, you can access them with the Help menu (Help => Log files => Show log files)

JawShoeAaa, 2022-04-09:
[2022-04-09 02-21-57.txt](https://github.com/SirLynix/obs-kinect/files/8456691/2022-04-09.02-21-57.txt)
like this then?

SirLynix, 2022-04-09:
Yes, and it shows the issue:
```
02:22:01.675: LoadLibrary failed for 'obs-kinect-azuresdk': The specified module could not be found.
02:22:01.675:  (126)
02:22:01.675: LoadLibrary failed for 'obs-kinect-freenect': The specified module could not be found.
02:22:01.675:  (126)
02:22:01.676: LoadLibrary failed for 'obs-kinect-freenect2': The specified module could not be found.
02:22:01.676:  (126)
02:22:01.677: LoadLibrary failed for 'obs-kinect-sdk10': The specified module could not be found.
02:22:01.677:  (126)
02:22:01.677: LoadLibrary failed for 'obs-kinect-sdk20': The specified module could not be found.
02:22:01.677:  (126)
```

How did you install obs-kinect? You're supposed to copy every file at the right location, not just the plugin folder.

JawShoeAaa, 2022-04-09:
i copied the two folders into my obs folder, how do i fix it?

SirLynix, 2022-04-09:
there are 3 folders: `bin`, `data` and `obs-plugins`, copy them to your obs installation

JawShoeAaa, 2022-04-09:
when i did that it found new files to copy, when i restarted OBS it works now

closed 2022-04-09 by SirLynix