SirLynix/obs-kinect #65: Azure Kinect only in RGB mode SDK Error - Depth Engine Not Found

lyfesaver74, 2022-04-25:
11:50:30.857: [obs-kinect] [azure] SDK error: dynlib_create(). Failed to load DLL depthengine_2_0 with error code: 193 (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\dynlib\dynlib_windows.c:115)
11:50:30.857: [obs-kinect] [azure] SDK error: deloader_init_once(). Failed to Load Depth Engine Plugin (depthengine). Depth functionality will not work (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\deloader\deloader.cpp:75)
11:50:30.857: [obs-kinect] [azure] SDK error: deloader_init_once(). Make sure the depth engine plugin is in your loaders path (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\deloader\deloader.cpp:76)
11:50:30.857: [obs-kinect] [azure] SDK error: deloader_depth_engine_create_and_initialize(). Failed to load depth engine plugin (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\deloader\deloader.cpp:113)
11:50:30.857: [obs-kinect] [azure] SDK error: depth_engine_start_helper(). Depth engine create and initialize failed with error code: 108. (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\dewrapper\dewrapper.c:154)
11:50:30.857: [obs-kinect] [azure] SDK error: deresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in depth_engine_start_helper() (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\dewrapper\dewrapper.c:160)
11:50:30.857: [obs-kinect] [azure] SDK error: depth_engine_start_helper(dewrapper, dewrapper->fps, dewrapper->depth_mode, &depth_engine_max_compute_time_ms, &depth_engine_output_buffer_size) returned failure in depth_engine_thread() (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\dewrapper\dewrapper.c:194)
11:50:30.857: [obs-kinect] [azure] SDK warning: capturesync_add_capture(). Capture Error Detected, Depth  (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\capturesync\capturesync.c:200)
11:50:30.857: [obs-kinect] [azure] SDK error: dewrapper_start(). Depth Engine thread failed to start (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\dewrapper\dewrapper.c:552)
11:50:30.857: [obs-kinect] [azure] SDK error: dewrapper_start(depth->dewrapper, config, depth->calibration_memory, depth->calibration_memory_size) returned failure in depth_start() (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\depth\depth.c:398)
11:50:30.858: [obs-kinect] [azure] SDK error: cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming() (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\depth_mcu\depth_mcu.c:359)
11:50:30.858: [obs-kinect] [azure] SDK error: depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063 (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\depth_mcu\depth_mcu.c:362)
11:50:30.858: [obs-kinect] [azure] SDK error: depth_start(device->depth, config) returned failure in k4a_device_start_cameras() (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\sdk\k4a.c:895)
11:50:30.858: [obs-kinect] [azure] Failed to start cameras!

SirLynix, 2022-04-25:
Hi, it looks like you're lacking [Microsoft Azure Kinect Depth Engine](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/depthengine.md) which you can get by installing the official Azure Kinect debian package.

Krayn, 2022-04-25:
he is on windows11 and the depthengine_2_0.dll (not exactly sure on the name) is installed as well as in his OBS/bin/64bit folder

lyfesaver74, 2022-04-25:
it's also in SDK folder 
[image]

lyfesaver74, 2022-04-25:
For some reason it started working. Thank you

lyfesaver74, 2022-04-25:
Actually, I spoke too soon. Now the RGB does not work, but depth does

SirLynix, 2022-04-25:
Could you check if you have the same issues with the official Azure Kinect Viewer? This looks like a hardware (USB perhaps) issue

lyfesaver74, 2022-04-26:
I got the RGB and Depth sorted.... I believe it was because I was using the 32bit when i need 64bit but cant be sure, either way, both RGB and Depth are working.

Where I am having an issue now is that the body stuff is not working.

[image]

I could not find the vcomp140.dll you said would be in the tools folder so i had to grab it from Windows/SYSWOW64 folder

SirLynix, 2022-04-26:
Did you [install the Azure Kinect Body Tracking SDK](https://github.com/SirLynix/obs-kinect#if-you-have-a-kinect-v3-azure-kinect)?

lyfesaver74, 2022-04-26:
Yes

[image]

As you can see not only is vcomp140.dll not in there it also has different file names for some of the files you are looking for I am guiessing because they are newer versions

lyfesaver74, 2022-04-26:
Both SDKs re in there
[image]

SirLynix, 2022-04-26:
Did you copy the following files to your obs-studio/bin/64bit folder?

- cublas64_100.dll
- cudart64_100.dll
- cudnn64_7.dll
- dnn_model_2_0.onnx
- k4abt.dll
- onnxruntime.dll
- vcomp140.dll

It's required for obs-kinect to work properly, it won't find them in your program files folder

lyfesaver74, 2022-04-26:
[image]

SirLynix, 2022-04-26:
That doesn't tell me if you just installed those files or copied them to your obs studio binary folder.

vcomp140.dll isn't the most important file there

lyfesaver74, 2022-04-26:
I copied all the fioles you asked for (but with different versions) and iI grabbed the vcomp form syswow64

lyfesaver74, 2022-04-26:
[image]

lyfesaver74, 2022-04-26:
I am using the most recent SDK, obviously, do I need to be using a specific one?

[image]

lyfesaver74, 2022-04-26:
I downgraded to 1.1.0 and vcomp is in there but the other files are still different names

[image]

SirLynix, 2022-04-26:
It looks like I missed an Azure Kinect Body Tracking SDK update, could you try using 1.0.1?

lyfesaver74, 2022-04-26:
wow, look at the size difference

[image]

Am trying 1.0.1 now

lyfesaver74, 2022-04-26:
Same result... but at this point i have copied so many different files in so many different places there could be version conflicts

lyfesaver74, 2022-04-26:
[image]

lyfesaver74, 2022-04-26:
[image]

lyfesaver74, 2022-04-26:
Deleted the old (actually newer but older installed) files that I could remember and still same result

[image]

lyfesaver74, 2022-04-26:
The fact that the release notes for Body SDK 1.1.0 specifically states that it now works with the Sensor SDK version 1.4.1 might mean that 1.0.1 will not work along side 1.4.1

[image]

Or are you using 1.4.1 + 1.0.1?

lyfesaver74, 2022-04-26:
I cant build a proper taco much less any code so I will need to wait on you, my friend :D

lyfesaver74, 2022-04-27:
Will the new build support all the newest SDKs or do I need to stay on Body DSK 1.0.1?

SirLynix, 2022-04-27:
You can use the Body SDK 1.1.1.

Here's a version of obs-kinect with Body Tracking SDK code enabled : https://github.com/SirLynix/obs-kinect/suites/6281103935/artifacts/224646231
Could you try it and tell me if it works for you? You'll still need Azure Body SDK dll in your obs bin folder

SirLynix, 2022-04-27:
Perfect! I'm glad it works with this version. Thank you for testing.

lyfesaver74, 2022-04-27:
Just a heads up, one of the early road blocks was this typo:

[image]

You are saying to grab the files from the Sensor SDK and some of those files are only found in the Body SDK folder. Also, the vcomp file is no longer located there

lyfesaver74, 2022-04-27:
[image]
[image]