SirLynix/obs-kinect #66: My Body Is Broken

lyfesaver74, 2022-04-28:
Like that title? :D LOL

Seriously tho... when I choose any of the options in background removal that contains "Body" the image just freezes until I switch it back to depth.

SirLynix, 2022-04-28:
I had the same issue when testing yesterday with only "Body" filtering, however "Body within Depth" and "Body or Depth" were working as expected.

Maybe the Kinect doesn't recognize you? Could you try to step back?
Do the Azure Kinect Body Tracking SDK examples work?

lyfesaver74, 2022-04-28:
I am not aware of any examples or how to use them. Pardon my lack of understanding :(

SirLynix, 2022-04-28:
Try to run the `C:\Program Files\Azure Kinect Body Tracking SDK\tools\k4abt_simple_3d_viewer.exe` executable to see if it works well.

lyfesaver74, 2022-04-28:
[image]

lyfesaver74, 2022-04-28:
It's not being used by any other application. In fact, the Azure Kinect sensor viewer, LIV and OBS are all able to access the camera

SirLynix, 2022-04-28:
This is weird, ensure all other applications are properly closed before running it, and maybe restart your computer if it still doesn't work.

lyfesaver74, 2022-04-28:
[image]

Fresh reboot and now, it pops up the viewer but then before showing anything other than the blank window it closes

Event Viewer:
[image]

lyfesaver74, 2022-04-28:
[image]

SirLynix, 2022-04-28:
It looks like there's a problem with this SDK version, maybe another version would work better. I can't test it right now however.

lyfesaver74, 2022-04-28:
I understand