SirLynix/obs-kinect #67: Plugin not working (probably because of VS 2022)

cefoot, 2022-05-30:
When trying to start anything depth related in obs (for example greenscreen)
The Kinect (Azure) turns off and the screen in obs turns black.

In the log I notice the error:

> 14:46:09.313: [obs-kinect] [azure] SDK error: dynlib_create(). Failed to load DLL depthengine_2_0 with error code: 126 (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\dynlib\dynlib_windows.c:115)
14:46:09.313: [obs-kinect] [azure] SDK error: deloader_init_once(). Failed to Load Depth Engine Plugin (depthengine). Depth functionality will not work (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\deloader\deloader.cpp:75)
14:46:09.313: [obs-kinect] [azure] SDK error: deloader_init_once(). Make sure the depth engine plugin is in your loaders path (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\deloader\deloader.cpp:76)
14:46:09.313: [obs-kinect] [azure] SDK error: deloader_depth_engine_create_and_initialize(). Failed to load depth engine plugin (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\deloader\deloader.cpp:113)
14:46:09.313: [obs-kinect] [azure] SDK error: depth_engine_start_helper(). Depth engine create and initialize failed with error code: 108. (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\dewrapper\dewrapper.c:154)
14:46:09.313: [obs-kinect] [azure] SDK error: deresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in depth_engine_start_helper() (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\dewrapper\dewrapper.c:160)
14:46:09.313: [obs-kinect] [azure] SDK error: depth_engine_start_helper(dewrapper, dewrapper->fps, dewrapper->depth_mode, &depth_engine_max_compute_time_ms, &depth_engine_output_buffer_size) returned failure in depth_engine_thread() (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\dewrapper\dewrapper.c:194)
14:46:09.313: [obs-kinect] [azure] SDK warning: capturesync_add_capture(). Capture Error Detected, Depth  (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\capturesync\capturesync.c:200)
14:46:09.313: [obs-kinect] [azure] SDK error: dewrapper_start(). Depth Engine thread failed to start (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\dewrapper\dewrapper.c:552)
14:46:09.314: [obs-kinect] [azure] SDK error: dewrapper_start(depth->dewrapper, config, depth->calibration_memory, depth->calibration_memory_size) returned failure in depth_start() (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\depth\depth.c:398)
14:46:09.319: [obs-kinect] [azure] SDK error: cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming() (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\depth_mcu\depth_mcu.c:359)
14:46:09.319: [obs-kinect] [azure] SDK error: depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063 (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\depth_mcu\depth_mcu.c:362)
14:46:09.322: [obs-kinect] [azure] SDK error: depth_start(device->depth, config) returned failure in k4a_device_start_cameras() (in C:\Users\runneradmin\AppData\Local\.xmake\cache\packages\2202\k\k4a\v1.4.1\source\k4a\src\sdk\k4a.c:895)
14:46:09.322: [obs-kinect] [azure] Failed to start cameras!

My guess is that there is a problem with the "Visual Studio 2019 redistributables" sadly I can't install it because C++ redistributables 2015-2022 are already installed.

SirLynix, 2022-05-30:
It looks like you're lacking the [Azure Kinect SDK](https://docs.microsoft.com/en-us/azure/kinect-dk/sensor-sdk-download) depth engine, could you try to download and install it from Microsoft website then try again?

cefoot, 2022-05-30:
Sadly that's not the case. 
I just checked SDK 1.4.1 is installed and working.
[image]

(just to be sure I removed the SDK and installed it again)
Last year I've often used your plugin (when I was still using VS2019).

SirLynix, 2022-05-30:
I don't think it has anything to do with Visual Studio version.
Could you try to copy the depthengine_2_0.dll from `C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\bin\` to your obs-studio/bin/64bit folder?

cefoot, 2022-05-30:
done... same error :/

closed 2022-05-30 by cefoot