SirLynix/obs-kinect #70: [Question] Possible to add support for Windows Hello webcams?

hanjo, 2022-07-27:
Hi,

this is a pretty cool project, thanks for all the work!

As far as I understand, webcams supporting Windows Hello are also equipped with an infrared camera and should be able to provide depth information. Given this, would it be technically possible to use those cameras as well?

SirLynix, 2022-07-27:
Hi!

Thanks for your interest in obs-kinect. As explained on the readme, this plugin could work with other devices as well, as long as someone writes a backend for the target device.

So supporting Windows Hello shouldn't be an issue, given someone with a Windows Hello camera makes such a backend. I would gladly do it but my only infrared camera are the Kinect ones.

GraphicHealer, 2025-03-06:
So I was actually curious about this myself at one point, but unfortunately it seems the only reason Windows Hello has an IR camera is to prevent issues with lighting, and give a more reliable 2D read of your face. There is no IR Projector, just an IR lamp, and the Hello cams do not give and cannot physically obtain any sort of depth mapping.

See here: https://learn.microsoft.com/en-us/windows-hardware/design/device-experiences/windows-hello-face-authentication