SirLynix/obs-kinect #71: v1.0 RC1 Linux install instructions

KeiranHines, 2022-08-10:
As with a few people in issue #9 I can't seem to get anything working on Ubuntu 20.04
```
error: os_dlopen(./obs-kinect-freenect2->./obs-kinect-freenect2.so): ./obs-kinect-freenect2.so: cannot open shared object file: No such file or directory
```

I have the latest build of libfreenect2 built and copied to /usr/lib/x86_64-linux-gnu/ and the obs-kinect plugin at /usr/lib/obs-plugins

Any assistance on this would be greatly appreciated. 

Log output is as follows 
```
12:47:04.387: FFMPEG VAAPI supported
12:47:04.388: os_dlopen(/usr//lib/obs-plugins/obs-kinect-azuresdk.so->/usr//lib/obs-plugins/obs-kinect-azuresdk.so): libk4a.so.1.4: cannot open shared object file: No such file or directory
12:47:04.388: 
12:47:04.388: Module '/usr//lib/obs-plugins/obs-kinect-azuresdk.so' not loaded
12:47:04.389: os_dlopen(/usr//lib/obs-plugins/obs-kinect-freenect.so->/usr//lib/obs-plugins/obs-kinect-freenect.so): libfreenect.so.0: cannot open shared object file: No such file or directory
12:47:04.389: 
12:47:04.389: Module '/usr//lib/obs-plugins/obs-kinect-freenect.so' not loaded
12:47:04.390: Failed to load 'en-US' text for module: 'obs-kinect.so'
12:47:04.390: os_dlopen(./obs-kinect-azuresdk->./obs-kinect-azuresdk.so): ./obs-kinect-azuresdk.so: cannot open shared object file: No such file or directory
12:47:04.390: 
12:47:04.390: os_dlopen(./obs-kinect-freenect->./obs-kinect-freenect.so): ./obs-kinect-freenect.so: cannot open shared object file: No such file or directory
12:47:04.390: 
12:47:04.390: os_dlopen(./obs-kinect-freenect2->./obs-kinect-freenect2.so): ./obs-kinect-freenect2.so: cannot open shared object file: No such file or directory
12:47:04.390: 
12:47:04.390: os_dlopen(./obs-kinect-sdk10->./obs-kinect-sdk10.so): ./obs-kinect-sdk10.so: cannot open shared object file: No such file or directory
12:47:04.390: 
12:47:04.390: os_dlopen(./obs-kinect-sdk20->./obs-kinect-sdk20.so): ./obs-kinect-sdk20.so: cannot open shared object file: No such file or directory
```

SirLynix, 2022-08-10:
The issue is that it depends on your distribution and how you installed OBS Studio, as far as I know.

How did you install OBS? Using snap or something else?

It looks like it finds obs-kinect-azuresdk.so and obs-kinect-freenect.so but can't load their dependencies, however there's nothing about obs-kinect-freenect2.so except it just doesn't find it, are they in separate folders?

KeiranHines, 2022-08-10:
Hi @SirLynix 
obs-kinect-freenect2 is in the load 5th last line 
```
12:47:04.390: os_dlopen(./obs-kinect-freenect2->./obs-kinect-freenect2.so): ./obs-kinect-freenect2.so: cannot open shared object file: No such file or directory
```
I am installed via apt and on Ubuntu 20.04

SirLynix, 2022-08-10:
Yeah I meant there's no error with freenect2 backend where it says it doesn't find its dependencies.
Could you try to put obs kinect backends .so files (so not obs-kinect.so) next to obs studio binary?

KeiranHines, 2022-08-10:
I ran a couple of tests and found some useful information. 

1: Copied obs-kinect-*.so to /usr/bin/ and ran obs from the shortcut.
    result: same error as above
2: changed directory to /usr/bin and ran obs
    result: Obs started and Kinect worked
3: copied the obs-kinect-*.so files to a temp directory /tmp/test to be specific, changed to that directory and ran obs
    result: Obs started and Kinect worked. 
    
based on that I would assume obs-kinect is hard coded to load from the current working directory rather than plugins or something similar.

SirLynix, 2022-08-10:
This may be because obs-kinect loads them using a relative path, I pushed a commit to try to fix this, could you please test with this version?

https://github.com/SirLynix/obs-kinect/actions/runs/2832257366

(you will be able to download the artifacts once the CI completes)

KeiranHines, 2022-08-10:
Worse result than before. 
Launching obs from the shortcut, or from a folder with the backend no-longer works.
```
09:01:18.454: FFMPEG VAAPI supported
09:01:18.455: os_dlopen(/usr//lib/obs-plugins/obs-kinect-azuresdk.so->/usr//lib/obs-plugins/obs-kinect-azuresdk.so): libk4a.so.1.4: cannot open shared object file: No such file or directory
09:01:18.455: 
09:01:18.455: Module '/usr//lib/obs-plugins/obs-kinect-azuresdk.so' not loaded
09:01:18.455: os_dlopen(/usr//lib/obs-plugins/obs-kinect-freenect.so->/usr//lib/obs-plugins/obs-kinect-freenect.so): libfreenect.so.0: cannot open shared object file: No such file or directory
09:01:18.455: 
09:01:18.455: Module '/usr//lib/obs-plugins/obs-kinect-freenect.so' not loaded
09:01:18.456: Failed to load 'en-US' text for module: 'obs-kinect.so'
09:01:18.456: os_dlopen(obs-kinect-azuresdk->obs-kinect-azuresdk.so): obs-kinect-azuresdk.so: cannot open shared object file: No such file or directory
09:01:18.456: 
09:01:18.456: os_dlopen(obs-kinect-freenect->obs-kinect-freenect.so): obs-kinect-freenect.so: cannot open shared object file: No such file or directory
09:01:18.456: 
09:01:18.456: os_dlopen(obs-kinect-freenect2->obs-kinect-freenect2.so): obs-kinect-freenect2.so: cannot open shared object file: No such file or directory
09:01:18.456: 
09:01:18.456: os_dlopen(obs-kinect-sdk10->obs-kinect-sdk10.so): obs-kinect-sdk10.so: cannot open shared object file: No such file or directory
09:01:18.456: 
09:01:18.456: os_dlopen(obs-kinect-sdk20->obs-kinect-sdk20.so): obs-kinect-sdk20.so: cannot open shared object file: No such file or directory
```

I will checkout the patch commit later today and see if I can build at test it later today or on the weekend. If I can get a working patch I will push it up for review.

KeiranHines, 2022-08-11:
Hi @SirLynix 
After some more testing I was able to find some form of solution. 

Using the commit you linked above I needed to put the backend libraries in /usr/lib/ not /usr/lib/obs-plugins
changing you commit to be 
```
#ifdef _WIN32
	ObsLibPtr lib(os_dlopen(path.c_str()));
#else
	ObsLibPtr lib(os_dlopen(("/usr/lib/obs-plugins/" + path).c_str()));
#endif
```
Also looked like it would work but my build was throwing out other issues due to issues compiling obs-studio so I was failing to launch the plugin at all. 
I suspect a dynamic path such as obs-plugins/ + path will work but I was unable to properly verify this. 

Either way. I am happy to just have the libraries at /usr/lib for now. if a better path such as /usr/lib/obs-plugins/ can be achieved I am happy to test it out.

SirLynix, 2022-08-11:
The big issue I have with this solution is that it won't work for everyone, it depends on your Linux distribution and how you installed OBS Studio.

Do you know about other OBS Studio plugins which have to load binaries like this? I wonder how they handle this.

KeiranHines, 2022-08-11:
I am not familiar with it so I did some research. 

Firstly os_dlopen wraps dlopen on linux. dlopen has a specified search order to look for libraries https://man7.org/linux/man-pages/man3/dlopen.3.html

Secondly looking at the obs-studio repo. os_dlopen is used just with the path [obs-studio example](https://github.com/obsproject/obs-studio/blob/e3f416f3fc72db49d1ba71b8423454d053a9d6bf/plugins/obs-ffmpeg/obs-ffmpeg.c#L262 )

As such I am happy to approve the change in the above commit as it complies with the OBS standard implementation. 
I would recommend a documentation update to make sure that is install instructions is clear to future readers.

SirLynix, 2022-08-11:
Thanks a lot for investigating.

So basically the plugins works fine on Linux if you move every backend .so in the /usr/lib folder?

Could you try with this version: https://github.com/SirLynix/obs-kinect/actions/runs/2841180426

I tried to set the rpath to the current folder and move the backend .so in the plugin folder, could you tell me if it's working better this way? (I don't have a working desktop Linux to test myself for now)

KeiranHines, 2022-08-13:
No such luck with the latest build. I tried with the backend at /usr/lib/obs-plugin and /usr/lib/obs. Both failed for me. It looks like there is an extra slash being injected into the path for the library. its logging /usr//lib/obs-plugins not /usr/lib/obs-plugins. See log below. 
 
 ```
15:11:13.415: os_dlopen(/usr//lib/obs-plugins/obs-kinect-azuresdk.so->/usr//lib/obs-plugins/obs-kinect-azuresdk.so): libobs-kinectcore.so: cannot open shared object file: No such file or directory
15:11:13.415: 
15:11:13.415: Module '/usr//lib/obs-plugins/obs-kinect-azuresdk.so' not loaded
15:11:13.415: os_dlopen(/usr//lib/obs-plugins/obs-kinect-freenect2.so->/usr//lib/obs-plugins/obs-kinect-freenect2.so): libobs-kinectcore.so: cannot open shared object file: No such file or directory
15:11:13.415: 
15:11:13.415: Module '/usr//lib/obs-plugins/obs-kinect-freenect2.so' not loaded
15:11:13.415: os_dlopen(/usr//lib/obs-plugins/obs-kinect-freenect.so->/usr//lib/obs-plugins/obs-kinect-freenect.so): libfreenect.so.0: cannot open shared object file: No such file or directory
15:11:13.415: 
15:11:13.415: Module '/usr//lib/obs-plugins/obs-kinect-freenect.so' not loaded
15:11:13.415: os_dlopen(/usr//lib/obs-plugins/obs-kinect.so->/usr//lib/obs-plugins/obs-kinect.so): libobs-kinectcore.so: cannot open shared object file: No such file or directory
15:11:13.415: 
15:11:13.415: Module '/usr//lib/obs-plugins/obs-kinect.so' not loaded
15:11:13.416: Failed to load 'en-GB' text for module: 'obs-libfdk.so'
```

SirLynix, 2022-08-13:
The issue here seems to be because of libos-kinectcore, which is common code for all obs kinect .so

Could you try again with this version?
https://github.com/SirLynix/obs-kinect/actions/runs/2851501403

KeiranHines, 2022-08-13:
With the backend in /usr/lib/obs-plugins/ the backends were not loading.
```
20:13:41.860: os_dlopen(/usr//lib/obs-plugins/obs-kinect-freenect.so->/usr//lib/obs-plugins/obs-kinect-freenect.so): libfreenect.so.0: cannot open shared object file: No such file or directory
20:13:41.860: 
20:13:41.860: Module '/usr//lib/obs-plugins/obs-kinect-freenect.so' not loaded
20:13:41.862: Failed to load 'en-US' text for module: 'obs-kinect.so'
20:13:41.862: os_dlopen(obs-kinect-azuresdk->obs-kinect-azuresdk.so): obs-kinect-azuresdk.so: cannot open shared object file: No such file or directory
20:13:41.862: 
20:13:41.862: os_dlopen(obs-kinect-freenect->obs-kinect-freenect.so): obs-kinect-freenect.so: cannot open shared object file: No such file or directory
20:13:41.862: 
20:13:41.862: os_dlopen(obs-kinect-freenect2->obs-kinect-freenect2.so): obs-kinect-freenect2.so: cannot open shared object file: No such file or directory
20:13:41.862: 
20:13:41.862: os_dlopen(obs-kinect-sdk10->obs-kinect-sdk10.so): obs-kinect-sdk10.so: cannot open shared object file: No such file or directory
20:13:41.862: 
20:13:41.862: os_dlopen(obs-kinect-sdk20->obs-kinect-sdk20.so): obs-kinect-sdk20.so: cannot open shared object file: No such file or directory
```
With the backend in /usr/lib. I was able to load the kinect and use it the same as the original release in this issue. 
```
20:16:40.457: os_dlopen(/usr//lib/obs-plugins/obs-kinect-freenect.so->/usr//lib/obs-plugins/obs-kinect-freenect.so): libfreenect.so.0: cannot open shared object file: No such file or directory
20:16:40.457: 
20:16:40.457: Module '/usr//lib/obs-plugins/obs-kinect-freenect.so' not loaded
20:16:40.457: Failed to load 'en-US' text for module: 'obs-kinect.so'
20:16:40.458: os_dlopen(obs-kinect-freenect->obs-kinect-freenect.so): libfreenect.so.0: cannot open shared object file: No such file or directory
20:16:40.458: 
20:16:40.460: os_dlopen(obs-kinect-sdk10->obs-kinect-sdk10.so): obs-kinect-sdk10.so: cannot open shared object file: No such file or directory
20:16:40.460: 
20:16:40.461: os_dlopen(obs-kinect-sdk20->obs-kinect-sdk20.so): obs-kinect-sdk20.so: cannot open shared object file: No such file or directory
```

o-kotb, 2022-11-15:
I'm trying to install it with the official obs flatpak. If it works there, it should work on all distributions. The plugin works, but it can't load libfreenect2 like here. With native installation (not flatpak) I could point to obs-plugins dir as the work dir and it would sometimes work (though crashes most of the time). If anyone knows how to set working directory in flatpak that would be helpful.

o-kotb, 2022-11-30:
continued in https://github.com/SirLynix/obs-kinect/issues/9#issuecomment-1332350668 with instructions

weirdal3333, 2022-12-29:
Update: I installed ubuntu bionic in a VM, followed the .yml file, and it fails the same way that it does on linux mint. Lots of steps and dependencies are missing from your instructions.