SirLynix/obs-kinect #76: (Azure Kinect V3) - Depth only clickable option for Faux Greenscreen

mattjohanek, 2023-01-08:
After adding the plugin to OBS and enabling the faux greenscreen option, I'm only able to click on the 'Depth' option provided. The other options 'Body, 'Body within depth', etc. are listed, but are unclickable and faded out. 

I've downloaded the latest 'Azure Kinect Body Tracking SDK' to my computer and copied the required files to the C:\Program Files\obs-studio\bin\64bit as mentioned in the "How to Use" section of the Readme.

Also, launching the 'k4abt_simple_3d_viewer' file in \Azure Kinect Body Tracking SDK\tools works well and detects my body.

I'd like to have the 'Body' options available with the faux greenscreen option. Has anyone encountered this problem with the Azure Kinect camera and been able to fix it?

 Any help or suggestions would be greatly appreciated.

Thanks for your time,

[image]

SirLynix, 2023-01-08:
Hi,

Could you send your obs logs here?

mattjohanek, 2023-01-08:
Sure,

Here is my latest log. Let me know if you need more information.

- Matt

13:11:14.908: CPU Name: 11th Gen Intel(R) Core(TM) i5-11400H @ 2.70GHz
13:11:14.908: CPU Speed: 2688MHz
13:11:14.909: Physical Cores: 6, Logical Cores: 12
13:11:14.909: Physical Memory: 16147MB Total, 10454MB Free
13:11:14.909: Windows Version: 10.0 Build 22621 (release: 2009; revision: 963; 64-bit)
13:11:14.909: Running as administrator: false
13:11:14.909: Aero is Enabled (Aero is always on for windows 8 and above)
13:11:14.909: Windows 10/11 Gaming Features:
13:11:14.909: 	Game DVR: On
13:11:14.909: 	Game Mode: Probably On (no reg key set)
13:11:14.909: 	Hardware GPU Scheduler: Probably On (no reg key set)
13:11:14.913: Sec. Software Status:
13:11:14.915: 	Norton Security Ultra: disabled (AV)
13:11:14.915: 	Microsoft Defender Antivirus: enabled (AV)
13:11:14.916: 	Norton Security Ultra: disabled (FW)
13:11:14.916: 	Windows Firewall: enabled (FW)
13:11:14.917: Current Date/Time: 2023-01-08, 13:11:14
13:11:14.917: Browser Hardware Acceleration: true
13:11:14.917: Hide OBS windows from screen capture: false
13:11:14.917: Qt Version: 6.3.1 (runtime), 6.3.1 (compiled)
13:11:14.917: Portable mode: false
13:11:15.190: OBS 28.1.2 (64-bit, windows)
13:11:15.190: ---------------------------------
13:11:15.799: ---------------------------------
13:11:15.799: audio settings reset:
13:11:15.799: 	samples per sec: 48000
13:11:15.799: 	speakers:        2
13:11:15.799: 	max buffering:   960 milliseconds
13:11:15.799: 	buffering type:  dynamically increasing
13:11:15.800: ---------------------------------
13:11:15.800: Initializing D3D11...
13:11:15.800: Available Video Adapters: 
13:11:15.811: 	Adapter 0: NVIDIA GeForce RTX 3050 Ti Laptop GPU
13:11:15.811: 	  Dedicated VRAM: 4169138176
13:11:15.811: 	  Shared VRAM:    4170870784
13:11:15.811: 	  PCI ID:         10de:25a0
13:11:15.811: 	  Driver Version: 31.0.15.2802
13:11:15.811: 	  output 0:
13:11:15.811: 	    name=
13:11:15.811: 	    pos={0, 0}
13:11:15.811: 	    size={1920, 1080}
13:11:15.811: 	    attached=true
13:11:15.811: 	    refresh=144
13:11:15.811: 	    space=RGB_FULL_G22_NONE_P709
13:11:15.811: 	    sdr_white_nits=80
13:11:15.811: 	    nit_range=[min=0.500000, max=270.000000, max_full_frame=270.000000]
13:11:15.811: 	Adapter 1: Intel(R) UHD Graphics
13:11:15.811: 	  Dedicated VRAM: 134217728
13:11:15.811: 	  Shared VRAM:    4170870784
13:11:15.811: 	  PCI ID:         8086:9a68
13:11:15.811: 	  Driver Version: 30.0.100.9864
13:11:15.819: Loading up D3D11 on adapter NVIDIA GeForce RTX 3050 Ti Laptop GPU (0)
13:11:15.959: D3D11 loaded successfully, feature level used: b000
13:11:15.959: DXGI increase maximum frame latency success
13:11:15.964: D3D11 GPU priority setup failed (not admin?)
13:11:17.115: ---------------------------------
13:11:17.115: video settings reset:
13:11:17.115: 	base resolution:   1920x1080
13:11:17.115: 	output resolution: 1280x720
13:11:17.115: 	downscale filter:  Bicubic
13:11:17.115: 	fps:               30/1
13:11:17.115: 	format:            NV12
13:11:17.115: 	YUV mode:          Rec. 709/Partial
13:11:17.115: NV12 texture support enabled
[... 62 lines cut ...]
13:11:17.567:     decklink-captions.dll
13:11:17.567:     coreaudio-encoder.dll
13:11:17.567: ---------------------------------
13:11:17.568: ==== Startup complete ===============================================
13:11:17.637: All scene data cleared
13:11:17.637: ------------------------------------------------
13:11:17.656: [win-wasapi: 'Desktop Audio'] update settings:
13:11:17.656: 	device id: default
13:11:17.656: 	use device timing: 1
13:11:17.659: [Loaded global audio device]: 'Desktop Audio'
13:11:17.660: [win-wasapi: 'Mic/Aux'] update settings:
13:11:17.660: 	device id: default
13:11:17.660: 	use device timing: 0
13:11:17.660: [Loaded global audio device]: 'Mic/Aux'
13:11:18.113: Switched to scene 'Scene'
13:11:18.117: ------------------------------------------------
13:11:18.117: Loaded scenes:
13:11:18.117: - scene 'Scene':
13:11:18.117:     - source: 'GS Pro View' (game_capture)
13:11:18.117:     - source: 'Kinect' (kinect_source)
13:11:18.117: ------------------------------------------------
13:11:18.441: WASAPI: Device 'Microphone Array (Realtek(R) Audio)' [48000 Hz] initialized
13:11:18.489: adding 42 milliseconds of audio buffering, total audio buffering is now 42 milliseconds (source: Mic/Aux)
13:11:18.489: 
13:11:18.867: WASAPI: Device 'Speakers (Realtek(R) Audio)' [48000 Hz] initialized

SirLynix, 2023-01-20:
I'm sorry I was very busy those last weeks, have you managed to fix this by yourself?

closed 2023-01-20 by mattjohanek

SirLynix, 2023-01-20:
I will take that as a yes