SirLynix/obs-kinect #78: Using Kinect source in OBS tanks performance in-game

pinecones-sx, 2023-02-04:
It took me a long time to track this down, but using a Kinect source in OBS while running some games (particularly Noita) will tank performance in the game.  Using another webcam or no webcam prevents any impact to game otherwise.

For Noita, this may have something to do with the game only being able to use one thread?  Not 100% sure.

Made sure everything was updated & reinstalled all related software with no improvement.

Computer specs:
- i9-10900kf cpu (3.70GHz, 10-core)
- GeForce RTX 2070 SUPER
- 16 GB DDR4 3200

SirLynix, 2023-02-04:
Hi,
What Kinect are you using?

pinecones-sx, 2023-02-04:
Sorry, that was dumb, the v2.

SirLynix, 2023-02-05:
The Kinect v2 is handled by a Windows service external to this plugin and ob, could you try to run a few experiments:

- try to lower the Kinect service priority (in the kinect source option)
- try to turn off and on the greenscreen effect (to see if it's the one causing this)
- try to run the game along with an official Kinect v2 sample (from https://www.microsoft.com/en-us/download/details.aspx?id=44561)

this will help to understand what the problem is and if I can do anything about it, obs-kinect already fetches frames from a separate thread.

pinecones-sx, 2023-02-06:
- Service priority was already set to lowest (Normal)
- disabling the green screen option seemed to greatly improve performance.  (not sure if it was still causing a minor impact)
- not 100% sure what you were looking for here but i installed the sdk and got a test app running that showed the depth and normal kinect camera views.  it appeared to have no impact on game performance.

SirLynix, 2023-02-06:
Alright so from what you're telling me the performance impact comes from the greenscreen effect, which seems a bit odd as you have a good GPU.
How many FPS do you have on OBS? What's the GPU usage with and without the greenscreen effect? (without playing any game)
Does checking/unchecking the "Use GPU to fetch color-to-depth ..." improves the usage?

pinecones-sx, 2023-02-06:
I'll see about the other items later, but checking the "use GPU" box didn't
improve performance.