SirLynix/obs-kinect #80: The AUR Package

barfin, 2023-03-18:
So I'm trying to create an AUR -bin package for this plugin but it causes obs to crash when i add the source in obs
here is the script
[PKGBUILD.txt](https://github.com/SirLynix/obs-kinect/files/11007097/PKGBUILD.txt)

here is the obs log:
[obslog.txt](https://github.com/SirLynix/obs-kinect/files/11007100/obslog.txt)

here is the file structure after installing the PKGBUILD
[image]

and here is the video of the crash:
[Screencast from 2023-03-18 04-00-08.webm](https://user-images.githubusercontent.com/37030489/226073829-19052136-08b7-4107-8c7f-155dfe6d5a65.webm)

SirLynix, 2023-03-18:
It looks like library files and shader files are not at the right location (aka where obs expects them), this is the cause of the crash.

Unfortunately, I really have no idea what the right location is, as it depends on the Linux distribution and package manager, which is the sole reason Linux is hard to support for this plugin.

barfin, 2023-03-18:
> It looks like library files and shader files are not at the right location (aka where obs expects them), this is the cause of the crash.
>
> Unfortunately, I really have no idea what the right location [...]

I'm not sure, for example this is the file structure for obs-vkcapture plugin and it works just fine, it's very similar to obs-kinect but this one works and obs-kinect doesn't
[image]

barfin, 2023-05-21:
I don't use arch linux anymore

closed 2023-05-21 by barfin