SirLynix/obs-kinect #81: Give the background blur option

Gamesnic, 2023-03-19:
Basically, if you've already seen it - a camera has a transition between the unfocused and focused stages. while the focus point is the most sharp, the sharpness falls off into the back and front. In your awesome implementation there's a single "zone" which is sharp, it'd be epic if you could change that.
Also, what about an option to set the focus point onto the face?

[image]

SirLynix, 2023-03-20:
This seems like a great idea, however what you're talking about is the focus point of the Kinect, I'm not sure I can parameter it on any version. See here for Azure Kinect: https://stackoverflow.com/questions/58304865/how-to-change-focus-of-azure-kinect-azure-kinect-sensor-sdk

This is a physical setting of the Kinect

Gamesnic, 2023-03-20:
Oh no no, not the physical focus point, I mean the virtual one. Basically so that the filter keeps the depth layer of the face sharp and blurs everything in the front and back. 
As if it was autofocus, but in software. We're simulating this bokeh afterall.

Gamesnic, 2023-03-20:
Yeah, basically. Maybe you could make it detect exclusively the head?
A real camera with smart AF normally uses the head as basepoint, that's why I was asking.
That together with realistic blur falling off behind and front would make it look quite like a professional DSLR.

SirLynix, 2023-03-20:
This seems like an interesting idea, I'm thinking about adding a "face" filter type which would allow for this kind of effect.