SirLynix/obs-kinect #82: Kinect lost connection every 5 seconds

HerrEckardt, 2023-03-20:
I used the Kinect for around a year and new there are problems with the connection. I tried every USB 3.x port from my PC, reinstalled all related drivers and drivers without any success. There weren't any issues with my PC from work. A friend of mine has the same unsolved issues.

SirLynix, 2023-03-20:
Hi, sorry to hear that.

What's the Kinect model? Could you try official Microsoft examples from the [corresponding SDK](https://github.com/SirLynix/obs-kinect#how-to-build-for-people-wanting-to-contribute) to see if the issue also happens?

OBS logs could also help

HerrEckardt, 2023-03-20:
Oh, I'm using the Kincet v2 (XBox One)

HerrEckardt, 2023-03-20:
The fan isn't spinning but the Kinect has a normal temperature and at my office PC from work it's working fine without any issue.

SirLynix, 2023-03-20:
If the fan isn't spinning it may still overheat quite easily, check this link: https://www.reddit.com/r/xboxone/comments/3piijf/psa_for_anyone_who_has_their_kinect_randomly_shut/

HerrEckardt, 2023-03-20:
I've connected the Kinect to my working PC and started the camera app from Windows. After a few minutes the fan starts spinning and the vent is clean. At my private PC the Kinect won't work even for a few minutes.

HerrEckardt, 2023-03-20:
I found the error and I'm feeling dumb now.
I deactivated the audio input "XBox NUI Sensor" in the sound setting from Windows a couple of days ago. After I activated it again the Kinect is working fine again.
Thanks for your help and sorry for my stupidity.

closed 2023-03-20 by HerrEckardt