SirLynix/obs-kinect #86: Crash on linux with egl errors

VectorKappa, 2023-10-16:
Upon opening the config window obs hangs and crashes.
```
[Info] [Freenect2DeviceImpl] starting...
error: Failed to make context current: EGL_SUCCESS
error: Failed to make context current: EGL_SUCCESS
error: Failed to make context current: EGL_SUCCESS
error: Failed to make context current: EGL_SUCCESS
error: Failed to make context current: EGL_SUCCESS
error: Failed to make context current: EGL_SUCCESS
error: eglSwapInterval failed
error: Cannot swap EGL buffers: EGL_BAD_SURFACE
error: Failed to make context current: EGL_SUCCESS
error: eglSwapInterval failed
error: Cannot swap EGL buffers: EGL_BAD_SURFACE
error: Failed to make context current: EGL_SUCCESS
error: eglSwapInterval failed
error: Cannot swap EGL buffers: EGL_BAD_SURFACE
error: Failed to make context current: EGL_SUCCESS
error: Failed to make context current: EGL_SUCCESS
error: Failed to make context current: EGL_SUCCESS
error: Failed to make context current: EGL_SUCCESS
error: Failed to make context current: EGL_SUCCESS
error: Failed to make context current: EGL_SUCCESS
error: eglSwapInterval failed
error: Cannot swap EGL buffers: EGL_BAD_SURFACE
error: Failed to make context current: EGL_SUCCESS
error: eglSwapInterval failed
error: Cannot swap EGL buffers: EGL_BAD_SURFACE
error: Failed to make context current: EGL_SUCCESS
error: eglSwapInterval failed
error: Cannot swap EGL buffers: EGL_BAD_SURFACE
error: Failed to make context current: EGL_SUCCESS
error: Failed to make context current: EGL_SUCCESS
error: Failed to make context current: EGL_SUCCESS
error: Failed to make context current: EGL_SUCCESS
error: Failed to make context current: EGL_SUCCESS
error: Failed to make context current: EGL_SUCCESS
error: eglSwapInterval failed
error: Cannot swap EGL buffers: EGL_BAD_SURFACE
error: Failed to make context current: EGL_SUCCESS
error: eglSwapInterval failed
error: Cannot swap EGL buffers: EGL_BAD_SURFACE
error: Failed to make context current: EGL_SUCCESS
error: eglSwapInterval failed
error: Cannot swap EGL buffers: EGL_BAD_SURFACE
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
info: [obs-kinect] [freenect2] Kinect active sources: ColorMappedDepth | Depth | Infrared
zsh: segmentation fault (core dumped)  obs
```
The error is DM-agnostic, happens both on wayland and X. obs-studio-tytan652 fork at least works, vanilla obs from repos crashes at startup.

```
OS: Arch Linux x86_64 
Kernel: 6.5.7-zen2-1-zen 
CPU: AMD Ryzen 5 3600 (12) @ 3.600GHz 
GPU: NVIDIA GeForce GTX 1660 SUPER 
Memory: 5430MiB / 15913MiB 
```

weirdal3333, 2024-07-29:
I think the devs have abandoned this project. It would have been nice if they finished what they started.

SirLynix, 2024-07-29:
Hi, sorry for not answering before and thanks for the reminder.

obs-kinect is feature stable and will still receive bugfixes, the problem here is that I can't really debug anything from this log. I can't even tell if the crash is coming from the plugin, from freenect or from OBS.

Also, since I'm not using it, Linux support is hard (distributing the plugin seems really messy compared to Windows). I'd be glad to have someone familiar to Linux help me on this part.

VectorKappa, 2024-07-29:
I'll try to get the stacktraces asap