SirLynix/obs-kinect #93: Compilation error (LibObs not found)

SirLynix, 2024-04-17:
Hi there, sorry I am an OBS noob. I have the following issue on my machines (mac and arch linux), I am assuming have skipped a very obvious step, please help.

```
obs-kinect on  master via 🌙 v5.4.6
❯ xmake build
error: ./xmake.lua:120: attempt to index a nil value (global 'LibObs')
stack traceback:
    [./xmake.lua:120]: in main chunk

checking for platform ... macosx
checking for architecture ... arm64
checking for Xcode directory ... %s
checking for Codesign Identity of Xcode ... no
error: ./xmake.lua:120: attempt to index a nil value (global 'LibObs')
stack traceback:
    [./xmake.lua:120]: in main chunk
```

_Originally posted by @djpiper28 in https://github.com/SirLynix/obs-kinect/issues/9#issuecomment-2060695394_

SirLynix, 2024-04-17:
@djpiper28 I moved your question here to try to keep #9 minimal.

However it seems you're missing the creation of config.lua, you need to duplicate [config.lua.default](https://github.com/SirLynix/obs-kinect/blob/master/config.lua.default) to config.lua and tell it where obs lies.

djpiper28, 2024-04-17:
> @djpiper28 I moved your question here to try to keep #9 minimal.
> 
> However it seems you're missing the creation of config.lua, you need to duplicate [config.lua.default](https://github.com/SirLynix/obs-kinect/blob/master/config.lua.default) to config.lua and tell it where obs lies.

This worked, I must have missed it in the README. Thanks for your help.

commit 0117c4bb (2024-04-17): Made build instruction clearer (fix #93)
closed 2024-04-17 by SirLynix