SirLynix/obs-kinect #94: Going from remove background to blur background causes crash, but going from replace>blur does not

Hauteknits, 2024-05-01:
System Info:

- Windows 11 Pro x64, Ryzen 5800X, X570 Chipset
- Kinect V2 using USB adapter, plugged into PCIe USB AiC with Renesis chipset, only device on controller
- OBS 29.0.0

I just got kinect-obs installed, and I was trying to use it as a better blur background. Selecting blur background would always trigger a crash when 'remove background' was previously selected (tried using both depth and body methods). While fiddling around with other settings, I discovered that if I use replace background first (empty file path) and then switch to blur background, it works flawlessly every time. Crash log attached
[Crash 2024-04-30 20-27-32.txt](https://github.com/SirLynix/obs-kinect/files/15173197/Crash.2024-04-30.20-27-32.txt)

SirLynix, 2024-05-01:
Thank you for the detailed info.

You're not the first person to create an issue about a crash with the background removal/blur with KinectV2, and unfortunately the report also points to a crash in the NVidia D3D10 driver.

Maybe you can try to use another API, or update your graphics drivers?

I will try to reproduce the crash again when I have some time.

Hauteknits, 2024-05-07:
Gotcha. I'm using the D3D11 renderer in OBS. I'll see what happens under using OpenGL, but I have a workaround at least that will allow me to use all 3 modes as stated above. I'll update my driver as well and see if that has any changes