SirLynix/obs-kinect #98: Kinect Source not appearing

SirLynix, 2025-05-14:
> Hi,
> I'm using the latest version of OBS (31.0.3) in Windows 10, and i can't see the Kinect Source option.
> 
> The Kinect itself is working correctly — all drivers are installed, and the Kinect Developer Toolkit displays the image just fine, SDK 1.8, Kinect V1.
> This seems to be caused by the Kinect plugin or its compatibility with the current OBS version. Is there an update or workaround available?
> 
> [image] 

 _Originally posted by @antetigmo in [#9](https://github.com/SirLynix/obs-kinect/issues/9#issuecomment-2879539228)_

antetigmo, 2025-05-14:
> Hello, I moved your issue here as it's not related to Linux.
> 
> Could you please post your OBS logs? Also check this part of the readme: https://github.com/SirLynix/obs-kinect?tab=readme-ov-file#i-copied-the-files-and-the-source-doesnt-show-up

Hey, thanks for the help and for moving the issue!

I'm heading to sleep now, and since I need the Kinect today, I'll probably take care of this later on maybe tomorrow, I'll send the logs and more details.

Also, I already did everything mentioned in that section.

Appreciate the attention!

antetigmo, 2025-05-16:
Hey, just a quick follow up, i ended up formatting the PC and it seems like it was a Windows issue after all.
Thanks again for the help and for taking the time to look into it.
Sorry for the trouble!

SirLynix, 2025-05-17:
Glad it finally worked, don't hesitate if you have other questions.

closed 2025-05-17 by SirLynix